philippoo66/optolink-splitter #13: List index out of range error when polling for items

FransOv, 2024-06-17:
Optolinkvs2-switch.py gives a List index out of range error a few seconds after starting. With polling disabled al goes well:
```
awaiting VS2...
Broker granted the following QoS: 0
VS detected
rx 06
rx 41 05 00 41 00 F8 01 3F
rx 41 0D 01 01 08 8E 08 AE 0B 55 00 AE 0B AE 0B 2D
list index out of range
exit close
closing serViCon
reset protocol
closing serViDev
cancel poll timer
exiting TCP/IP client
disconnect MQTT client
closing vitolog
```
[settings_ini.txt](https://github.com/user-attachments/files/15866492/settings_ini.txt)

I noticed in vitolog that the ACK is treated as part of the message from the slave. But I don't know whether that is relevant..
```
M	1718614049118	00
S	1718614049128	06
M	1718614049165	41 05 00 C1 00 F8 01 BF
S	1718614049217	06 41 06 01 C1 00 F8 01 20 E1
M	1718614049317	41 05 00 E1 00 F9 01 E0
S	1718614049370	06 41 06 01 E1 00 F9 01 4B 2D
M	1718614049470	41 05 00 01 00 FA 01 01
S	1718614049522	06 41 06 01 01 00 FA 01 01 04
M	1718614049622	41 05 00 21 00 FB 01 22
S	1718614049674	06 41 06 01 21 00 FB 01 46 6A
M	1718614049775	41 05 00 41 00 FC 01 43
S	1718614049827	06 41 06 01 41 00 FC 01 00 45
M	1718614049927	41 05 00 61 00 FD 01 64
S	1718614049979	06 41 06 01 61 00 FD 01 00 66
M	1718614050079	41 05 00 81 00 FE 01 85
S	1718614050131	06 41 06 01 81 00 FE 01 05 8C
M	1718614050231	41 05 00 A1 00 FF 01 A6
S	1718614050283	06 41 06 01 A1 00 FF 01 0F B7
```

philippoo66, 2024-06-17:
I realized the problem: your Vitocal uses another datetime format than the Vitodens:
```
rx 41 0D 01 01 08 8E 08 AE 0B 55 00 AE 0B AE 0B 2D

41 0D : STX, 13 Bytes Payload
    01 01 : Antwort auf virtual read request
        08 8E : DP Adresse
           08 : 8 bytes data:
AE 0B 55 00 AE 0B AE 0B : this is a different format. the 'Index out of Range' appears at
 
    weekdays = ['Mo', 'Di', 'Mi', 'Do', 'Fr', 'Sa', 'So']
    wkd = weekdays[int(data[4]) - 1]
   
 data[4] is AE, weekdays[0xAD] is out of range...
```
replace `    ("Anlagenzeit", 0x088E, 8, 'vdatetime'),`
by `    ("Anlagenzeit", 0x088E, 8, 'raw'),`

probably it is a Linux timestamp, we'll have to see...

FransOv, 2024-06-17:
I'll first check what datapoints I want to get from the Vitocal. I think the datapoint codes are not compatible between Vitodens and Vitocal.
Thanks for the superfast reply. It's a very nice piece of software you created. It is exactly what I needed. I have tried to interface via the KM-Bus with an ESP32, but could only sniff the transfer of data and never really succeeded in controlling the Vitocal.

FransOv, 2024-06-17:
Thanks, That saves a lot of time. I was just looking into decoding the Vitosoft xml files with https://github.com/sarnau/InsideViessmannVitosoft.

FransOv, 2024-06-17:
Just got the first real piece of data out of the Vitocal:
Message 8 received on vitocal/AussenTemp at 4:01 PM:
18.5
QoS: 0 - Retain: false

philippoo66, 2024-06-17:
> was just looking into decoding the Vitosoft xml files with

this is where the txt list is coming from. the xls was drived differently.

FransOv, 2024-06-17:
The xml file from Vitosoft of my system in excel format has the correct addresses for these datapoints:
```
    ("Buitentemperatuur", 0x0101, 2, 0.1, True),
    ("HK1_normale_temperatuur", 0x2000, 2, 0.1, False),
    ("HK1_verlaagde_temperatuur", 0x2001, 2, 0.1, False),
    ("HK1_temperatuur", 0x0116, 2, 0.1, False),
    ("HK2_normale_temperatuur", 0x3000, 2, 0.1, False),
    ("HK2_verlaagde_temperatuur", 0x3001, 2, 0.1, False),
    ("HK2_temperatuur", 0x0117, 2, 0.1, False),

Message 477 received on vitocal/HK2_temperatuur at 5:03 PM: 20
Message 476 received on vitocal/HK2_verlaagde_temperatuur at 5:03 PM: 17
Message 475 received on vitocal/HK2_normale_temperatuur at 5:03 PM: 19
Message 474 received on vitocal/HK1_temperatuur at 5:03 PM: 20.4
Message 473 received on vitocal/HK1_verlaagde_temperatuur at 5:03 PM: 17
Message 472 received on vitocal/HK1_normale_temperatuur at 5:03 PM: 20
Message 471 received on vitocal/Buitentemperatuur at 5:03 PM: 18.7

```
[ecnEventType VT200WO1C.xlsx](https://github.com/user-attachments/files/15873113/ecnEventType.VT200WO1C.xlsx)

philippoo66, 2024-06-18:
thank you for sharing the data points list! how did you get it? For the DP_...txt lists I used the python script of Markus 'sarnau' (PrintEventsForDatapoint.py?) with an 'updated' xml (ecnEventType.xml?), but the result seem not to be always correct?!?

FransOv, 2024-06-18:
I downloaded the Vitosoft program and connected the program to the vitocal by means of the Vitoconnect cable. The Excel is created by opening the ecnEventType.xml in the config directory of Vitosoft  in Excel. It'sthat simple.

philippoo66, 2024-06-18:
... aside that you need to install the Vitosoft monster with all its servers etc ...

I think it made an update when running Vitosoft? can you pls have a look at DPDefinitions.xml? mine is 186.559.004 Bytes, dated 14. ‎Dezember ‎2017, ‏‎11:44:00. Is your one different?

philippoo66, 2024-06-18:
I think the ecnEventType.xml is general. Look e.g. for outside temperature - you will find multiple addresses, depending on the device. Markus 'sarnau' described (more or less...) how to find the addresses according to the device

philippoo66, 2024-06-18:
another petition... Could you please have a look if there is a possibility in Vitosoft to remotely reset an fault/error?! Some guy is just looking for that...

FransOv, 2024-06-19:
_Could you please have a look if there is a possibility in Vitosoft to remotely reset an fault/error?!_ I don't see any. Vitosoft is meant to be used directly connected to the device so all resetting will have to be executed on the device itself. It does have the possibility to alter all parameters and maybe some would result in a reset, but I don't know which that would be.
_mine is 186.559.004 Bytes, dated 14. ‎Dezember ‎2017, ‏‎11:44:00. Is your one different?_ My ecnDataPointType.xml is dated 10-4-2024, the size  is exactly the same as your version.

philippoo66, 2024-06-19:
> Vitosoft is meant to...

you are right. Thank you anyway, also for looking at the xml!

FransOv, 2024-06-19:
I'll try to extract the information directly out of the Vitosoft database. It will take some time, as there are a lot of tables (100+) in the database.

philippoo66, 2024-06-19:
I also got me the SQL DB and made a 'screenshot' with access (for me this is much simpler to create queries etc and the DB is not 'alive' so not changing (regarding the main/important data). If you like I could upload the accdb (including my already generated queries) somewhere for download?

FransOv, 2024-06-20:
I'm using Ms SQL Management Studio to access a copy of the database. That works quite well. I'm using this query to get the relevant parameters for my Vitocal 300G:
```
select dp.Address as Model, 
et.Address as Parameter, et.Conversion, et.DefaultValue, 
ev.Unit, ev.Stepping, ev.DataType, ev.ValuePrecision, ev.LowerBorder, ev.UpperBorder, ev.Description
from ecnDeviceTypeDataPointTypeLink dev 
join ecnDatapointType dp on dev.DataPointTypeId=dp.Id 
join ecnDataPointTypeEventTypeLink dpe on dp.Id=dpe.DataPointTypeId
join ecnEventType et on et.id=dpe.EventTypeId
join ecnEventTypeEventValueTypeLink etvl on etvl.EventTypeId=et.Id
join ecnEventValueType ev on etvl.EventValueId=ev.Id

where dp.Address='CU401B_G'
```
This gives me most of the info. I still need to find the location info  (block length, byte pos, byte length, bit pos bit length). The info is not 100% complete nor 100% right. E.g. Outside temperature is given as 0x01C1 while it really is 0X0101.

FransOv, 2024-06-20:
On the question about the remote reset, these could possibly do. I have not tried any of them:
```
WPR3_BedienparameterHK1FunktionReset~0xA051	0xA051
WPR3_Konfi_FunktionNeustart~0xA050	                        0xA050
WPR3_Reset_Parameter_ALZ~0xA050	                        0xA050
WPR3_BedienparameterHK2FunktionReset~0xA051	0xA051
WPR3_BedienparameterHK3FunktionReset~0xA051	0xA051

```

philippoo66, 2024-06-20:
thank you! even in that case it is an old Vitodens, I think we have to check for this explicitly...

FransOv, 2024-06-21:
I see 4 entries for the Vitodens in the database. I can run the query for a one of them If you need it.
```
Technische Produktbeschreibung: ab Softwareindex 200, Projekt Vitodens 300 mit HO2B
Technische Produktbeschreibung: ab Softwareindex 200, Projekt Vitodens 200 mit HO1B
Technische Produktbeschreibung: ab Softwareindex 200, Projekt Vitodens 200 mit HO1B ab 04/2016
Technische Produktbeschreibung: ab Softwareindex 200, Projekt Vitodens 300 mit HO2B, ab 08/2016 mit Lüftungsbedienung
```

closed 2024-06-23 by FransOv