philippoo66/optolink-splitter #19: Error descriptions with return code 3

FransOv, 2024-07-18:
I have some trouble writing datapoints that according to the documentation are writable, but return an error: w ;0x3022;2;22 returns the response:3;0x3022;21 
The 3 means Error message, presumably identified by the returned data 21. Any idea what the different error messages are?

philippoo66, 2024-07-18:
Hi Frans!

In fact I never tested writing of 2 byte 'decimals' since there is no such 2 byte value in Vitodens to be written. The 'multi-byte values' such as time schedules have to get written 'top-down' (MSB-LSB - kind of big endian, even each time is only one byte as far as I remember). 

So it is possible that 2 byte decimals need to get written LSB-MSB, what would make sense since when reading they come LSB-MSB as well. So I think I should apply your change. THANK YOU very much!

>Turns out that with raw writes you do not specify the length as it is implicit in the hex string. The description of the raw writes in the wiki is incorrect.

Pls have a look at line 166 of request_util:
`            bval = (utils.get_int(parts[3])).to_bytes(int(parts[2]), 'little')`
First the value (parts[3]) is converted to integer, what means that decimal or hex (leading 0x) can be given. 
Then this gets converted to_bytes with a lenght of parts[2], which is the data length according to the wiki.
These bytes (with according/specified length) get performed by write_datapoint_ext, which takes the length from the len(bytes)
  
I think this is correct, or what do you think? I think the only bug was the wrong endian?

last not least: I do not have any information nor even an idea of the meaning of the error codes returned with a rejected request - sorry!

philippoo66, 2024-07-18:
SORRY! The line 166 apllies to 'write', not 'writeraw'. YOU ARE RIGHT, I will correct the Wiki, THANKS AGAIN!

FransOv, 2024-07-19:
I searched through the code of a number of Optolink projects, but none of them had a list of error codes. At least I'm fairly sure now that error  code 21 is "Value out of bounds".

philippoo66, 2024-07-19:
perhaps it would make sense to crawl Vitosoft for the codes? 
As far as I can see this early morning, the 21 is hex, isn't it?

FransOv, 2024-07-19:
Looks like hex yes. Lines 170-172:
```
            elif(data):
                # probably error message
                val = utils.arr2hexstr(data)  #f"{int.from_bytes(data, 'little')} ({utils.bbbstr(data)})"
```

philippoo66, 2024-07-19:
do you have posibility to decompile Vitosoft.exe? My current Visual Studio is too old (2015), I think the later ones can do

FransOv, 2024-07-19:
You could try to use the strings command of Windows to find strings in an executable. However, I think the Vitosoft program stores it strings externally to support the translations. I'm looking at the Textresource_en.xml file in the ServiceTool\Web\XmlDocuments\ subdirectory of the program files directory of the Vitosoft program. Looks promising, but I have not found it yet.

philippoo66, 2024-07-19:
problem with text resource may be that the code mapping may be missing...

FransOv, 2024-07-20:
I cannot find the error codes in the text resource xml. 
However by testing I found:
0x01: invalid address
0x04: invalid length
0x21: value out of bounds
writing a read only value is ignored, does not return an error.

closed 2024-07-22 by FransOv