philippoo66/optolink-splitter #22: Use optolink-splitter with the Vitosoft program

FransOv, 2024-07-27:
I've been trying to use serlog.py to log the communication between the Vitosoft program and my Vitocal heatpump. It does not work the vs2 protocol is not detected nothing is written to the log and the connection between Vitosoft and Vitocal is not established. Also using optolinkvs2_switch.py with the Vitosoft program  will not create a connection. If I un-comment the print of the ring buffer in serlog.py, all I see is:
```
7D FB FB
FB FB FB
FB FB FB
FB FB FB
7D FB FB
FB FB FB
FB FB FB
7D 7D 7D
FB FB FB
7D FB FB
FB FB 7D
7D 7D 7D
7D 7D 7D
7D 7D 7D
FB FB FB
FB FB FB
FB FB FB
7D 7D 7D
7D 7D 7D
7D 7D 7D
FB FB FB
7D 7D 7D
```
In the log of the Vitosoft program I see it is continuously sending 0x04.
Any ideas what can cause this?

philippoo66, 2024-07-27:
vs2 is not really necessary, I biult the detection only for initial purposes. it has no influence on the passing-through of the data.

I think something with the connection settings is wrong.

If something gets passed to the buffer, then fdata flag _must_ get set true and further down there _must_ something get written to the log.

- could you pls check if everything is ok with the system, reboot the optolink target and stuff like this?!
- if you use a Raspi higher than 2, the tty0 is usable only once a livetime with parity unequal none (see [here](https://forums.raspberrypi.com/viewtopic.php?t=289592#p1750695)) but this schould give an error message as far as I remember
-

FransOv, 2024-07-28:
Sorry, Was a missing driver for the CP2102.

closed 2024-07-28 by FransOv