philippoo66/optolink-splitter #30: Setting Available and Unavailable via MQTT

FransOv, 2024-09-05:
As it is now, whenever Optolink-Switch stops, whether intentionally or because the RasPi or the service encounters a problem, the latest values are displayed in Hom Assistant as if nothing is amiss. 
MQTT has the "Last Will and Testament" (LWT) functionality for this. In short this defines a message to be sent when the connection between the MQTT broker and the MQTT clients is lost or interrupted for some time and when it is started up again.
It would be nice if this LWT functionality was added to Optolink-Switch.
If you don't have time for this I'm willing to give it a try.

philippoo66, 2024-09-06:
sounds like a good idea! since you seem to have alread dealt with, it would be nice if you could implement! to me it is currently not clear how to trigger that last will when te process gets killed.

FransOv, 2024-09-07:
I'll try it next week. It looks rather simple, set a few parameters at connect and publish a message in the on_connect and on_disconnect callbacks.

FransOv, 2024-09-08:
I made the changes to mqqt_util.py (only three lines)  and created a pull request. I also changed the py program in the Wiki on Home_assistant that generates the Home Assistant entities to include an availability_topic for each sensor and control.

FransOv, 2024-09-11:
Pull request https://github.com/philippoo66/optolink-splitter/pull/32 submitted

closed 2024-09-11 by FransOv

philippoo66, 2024-09-11:
great, thank you. merged. 

it would be great if you could usually work with the develop branch because I'm afraid now the develop branch is behind main. As far as I remember main was up to date witch develop, so we can merge main into develop, but we need to compare...