philippoo66/optolink-splitter #34: Service does not run after reboot when only connected via WiFi

FransOv, 2024-09-10:
The Service description specifies: After=network.target
However this alsways does not ensure conncectivity in case of a WiFi connection, only that the network stack is started. Even After=network-online.target does not do this. It only means to wait untill the netwrok interface is configured correctly and has an IP address. In case of a fixed IP address and a WiFi connection that can still mean that the network resources, particulary the MQTT server is accessible. In my case I noticed that the service always failed after a reboot because of MQTT server not accessible.
For some reason, I was not able to find out what, also having a [Timer] section with an OnBootSec did not work. I finally solved it by including a ExecStartPre=/bin/sleep 30 in the service section. 
I'd recommend adding The ExecStartPre line to the top of the service section of the example service definition file in page 120 of the Wiki. Like This:
[Service]
**ExecStartPre=/bin/sleep 30**
Type=simple

philippoo66, 2024-09-11:
hi Frans,

thank you again! I agree (adding the entry). can you please do it!? There are two 'autostart instructions, https://github.com/philippoo66/optolink-splitter/wiki/510-error:-externally%E2%80%90managed%E2%80%90environment-%E2%80%90%E2%80%90-venv also.

Do we have to drop a small note rgarding the background of this setting?

greetings!
Phil

FransOv, 2024-09-11:
Service definitions on page 120 and 510 changed.

closed 2024-09-11 by FransOv