philippoo66/optolink-splitter #37: Service not running

Rustimation, 2024-09-23:
one step at a time:
optolinkvs2_switch works perfectly when started in the foreground, which of  course is only good for testing purposes.

I followed the wiki instructions @ https://github.com/philippoo66/optolink-splitter/wiki/120-optolinkvs2_switch-automatisch-starten
I adjusted the path to /home/pi/optolink, user is pi.
However, I cannot start the service correctly

sudo systemctl status optolinkvs2_switch has the following output:
● optolinkvs2_switch.service
     Loaded: loaded (/etc/systemd/system/optolinkvs2_switch.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2024-09-23 17:09:34 CEST; 1s ago
    Process: 787 ExecStartPre=/bin/sleep 30 # Network online target does not ensure actual connectivity when using WiFi so wait 30 sconds (code=exited, status=1/FAILURE)
        CPU: 10ms

Sep 23 17:09:36 OptoLinkPi systemd[1]: /etc/systemd/system/optolinkvs2_switch.service:1: Assignment outside of section. Ignoring.
Sep 23 17:09:36 OptoLinkPi systemd[1]: /etc/systemd/system/optolinkvs2_switch.service:2: Assignment outside of section. Ignoring.
Sep 23 17:09:36 OptoLinkPi systemd[1]: /etc/systemd/system/optolinkvs2_switch.service:3: Assignment outside of section. Ignoring.
~

I have tried removing the wait statement but to no avail

Any idea why this is happening?

philippoo66, 2024-09-23:
if you have it installed using a virtual environment, that has to be referred. 

see 
https://github.com/philippoo66/optolink-splitter/wiki/510-error:-externally%E2%80%90managed%E2%80%90environment-%E2%80%90%E2%80%90-venv
for how to do that.

philippoo66, 2024-09-23:
ps. also the in-line comment may cause error as I was told. I remove that from the script template.

philippoo66, 2024-09-23:
@Rustimation  please let me know if this solved the issue an if we can close it!

Rustimation, 2024-09-24:
Perfect. Now it works like a charm. I should have read the above mentioned page alle the way down... and/or I seem to have overlooked the link to that page on wiki page 120-...
I will now change my Viessmann API centric Node-Red logic to work with the MQTT values provided by the Optolinnk-Splitter and document my experience in my [blog ](https://rustimation.eu)
Kind regards and many thanks for your help.
Chris

philippoo66, 2024-09-24:
great, thank you for the feedback! I'm looking forward to Optolink-Splitter getting mentioned in your blog :-)