philippoo66/optolink-splitter #44: Vitocal 200s (CU401B_S) auslesen

timbrd, 2024-10-04:
Hallo,

seit einiger Zeit funktioniert das Auslesen einiger Stromverbrauchswerte meiner Vitocal 200s. @philippoo66, nochmal danke für deine Hilfe!

Zwischenzeitlich habe ich mit open3e auch meine PV-Anlage, den Stromspeicher und Energy-Meter angebunden.

Ich möchte mich nun wieder der Wärmpepumpe widmen und weitere Daten auslesen. Hierfür habe ich mir aus dem openv-Wiki eine Liste von Adressen (DP-Listen) heruntergeladen. Für meine Vitocal gibt es auch eine entsprechende Datei (CU401B_S).

Einige Daten konnte ich damit auch sauber auslesen, bei anderen habe ich aktuell noch Probleme.
Beispielsweise erhalte ich bei den Temperaturangaben für Warmwasser nur unsinnige Werte.

Folgendes steht in der DP-Liste:
```
- WW Temperatur Oben (7714) [WPR3_WW_Temperatur_Oben~0x01CD (SInt)] HIDDEN:("WW Temperatur Oben"="Sensor nicht vorhanden")
- WW Temperatur Mitte (7718) [WPR3_WW_Temperatur_Mitte~0x01CF (SInt)] HIDDEN:("WW Temperatur Mitte"="Sensor nicht vorhanden")
- WW Temperatur Unten (7716) [WPR3_WW_Temperatur_Unten~0x01CE (SInt)] HIDDEN:("WW Temperatur Unten"="Sensor nicht vorhanden")
- WW Temperatur Ausgang (7720) [WPR3_WW_Temperatur_Ausgang~0x01D0 (SInt)] HIDDEN:("WW Temperatur Ausgang"="Sensor nicht vorhanden")
- (6000) Warmwassertemperatur-Sollwert (7142) [WPR_WW_Temp_soll~0x6000 (SInt)]
```

In meiner settings._ini.py:
```
("WW Temperatur Oben (01CD)", 0x01CD, 3, 0.1, True),
("WW Temperatur Unten (01CE)", 0x01CE, 3, 0.1, True),
("WW Temperatur Mitte (01CF)", 0x01CF, 3, 0.1, True),
("WW Temperatur Ausgang (01D0)", 0x01D0, 3, 0.1, True),
("WW Temperatur Soll (6000)", 0x6000, 3, 0.1, True),
```

Die ausgelesenen Werte:
```
WW Temperatur Oben (01CD): 4194477.0
WW Temperatur Unten (01CE): 458752.0
WW Temperatur Mitte (01CF): 39321.6
WW Temperatur Ausgang (01D0): 6598.6
```

Setze ich den Datentyp bzw. die Länge in der settings_ini.py auf "2", dann wird gar kein Wert zurückgegeben. In [diesem](https://github.com/openv/openv/issues/480#issuecomment-1364274127) Thread hier wurde dies ebenfalls angemerkt, die Frage wurde aber leider nicht beantwortet.

Könnte mir jemand unter die Arme greifen und mir einen Hinweis geben, wie ich hier weiterkomme?

philippoo66, 2024-10-04:
moin Tim!

3 ist eine sehr ungewöhnliche Länge, das hab ich noch nicht gesehen. SInt steht ja nun auch eindeutig für 2. Nur weil Werte kommen, wenn man eine unsinnige Länge ausliest, heisst das nicht dass die Werte irgendeinen Sinn ergeben. 

Hast du denn mehrere Sensoren im Speicher?

Hast du mal einfach die AD's ausgelesen (0x0800/802/804/.../80A/80D, alles 2 Bytes 1/10 Grad)? Vielleicht kommst du damit ja weiter... Bei mir ist 804 der (einzige) Speichersensor.

timbrd, 2024-10-04:
> moin Tim!
>
> 3 ist eine sehr ungewöhnliche Länge, das hab ich noch nicht gesehen. SInt steht ja nun auch eindeutig für 2. Nur weil Werte kommen, wenn man eine unsinnige Länge ausliest, heisst das [...]

Tatsächlich erhalte ich bei 0x0882 einen Wert (49.6), der meiner Warmwassertemperatur entspricht (ich beobachte noch). Allerdings habe ich im Haus zwei Warmwasserkreise und von 0800-08F8 finde ich den Wert für den zweiten Kreis nicht.
Wenn 0882 tatsächlich die Temperatur eines Warmwasserkreises angibt, ist dann die DB-Liste falsch, die ich verwende? Oder sind sie grundsätzlich fehlerbehaftet?

philippoo66, 2024-10-06:
die Listen sind jedenfalls grundsätzlich unvollständig. mit falsch oder nicht fällt mir schwer zu sagen wegen den ganzen 'HIDDEN's. Hatte @FransOv nicht schon eine Tabelle für deine Anlage generiert? Oft sind seine ja etwas treffender...

FransOv, 2024-10-06:
Temperature parameters of length 3 consist most of the times of 2 bytes  temperature followed by a 1  byte sensor status. you can use the byte-bit filter to extract the temperature.

philippoo66, 2024-10-16:
@timbrd kannst du noch mal die Werte mit 3 bytes als hex auslesen?!

timbrd, 2024-10-17:
> @timbrd kannst du noch mal die Werte mit 3 bytes als hex auslesen?!

Hi Philipp,

ich habe noch nicht verstanden, wie ich den [ByteBit Filter](https://github.com/philippoo66/optolink-splitter/wiki/010-Command-Syntax#write-raw-requests-hex-byte-string) verwende.

````
("WW Temperatur Oben (01CD)", 0x01CD, 3, 'b:0:2::raw'),
````

Als Wert erhalte ich dann "af0140"?

FransOv, 2024-10-17:
Ich lese die Temperatur in eine 3 byte Parameter so: ("primary_supply_temperature", 0xB400, 3, 'b:0:1', 0.1, True),

philippoo66, 2024-10-17:
moin Tim!

zum raw/hex auslesen hätte `read;0x01CD;3` bzw. `("WW Temperatur Oben (01CD)", 0x01CD, 3)` gereicht.

mit `'b:0:2::raw'` stellst du byte 0 bis 2 als raw dar, was auf's Gleiche rausläuft.

Wenn du jetzt die ersten beiden Bytes als vorzeichenbehaftete Dezimalzahl (x0,1) haben möchtest, wäre
`("WW Temperatur Oben (01CD)", 0x01CD, 3, 'b:0:1', 0.1, True)` das Richtige.

mit `'b:<first>:<last>'` filterst du ein Byte-Sequenz aus dem gelesenen Wert, `'b:0:1'` nimmt also die ersten beiden Bytes. 

Du kannst dann noch eine Bit-Maske über den Wert legen, um auf bestimmte Flags zu filtern. Angenommen, 4. Bit (Bit 3 von 0 aus -> 2^3=8) im 3. Byte stände für 'Sensor hat Kurzschluss', und das 5 Bit (Bit 4 von 0 aus -> 2^4=16) für 'Sensor Leitungsunterbrechung':
`("WW Temperatur Oben (01CD), Sensor Fehler", 0x01CD, 3, 'b:2:2:24', 1)`

Im Falles einen Sensorkurzschlusses würde das dann 8 liefern, bei Leitungsunterbrechung 16 und bei keinem von beidem 0, egal was in den anderen 6 Bits steht.

Wenn du 
```
("WW Temperatur Oben (01CD)", 0x01CD, 3, 'b:0:1', 0.1, True),
("WW Temperatur Oben (01CD), Sensor Fehler", 0x01CD, 3, 'b:2:2:24'),
```
direkt hintereinander schreibst, wird dabei der Datenpunkt nur 1 mal gelesen und mehrfach ausgewertet.

Kommen wir jetzt zu deinem Wert af0140:

die ersten beiden Bytes: af 01, litte endian -> 0x01af -> dez 431 -> 43,1°C - könnte das passen?
das käme heraus mit `("WW Temperatur Oben (01CD)", 0x01CD, 3, 'b:0:1', 0.1, True),`

das letzte Byte 40 -> binär 01000000 -> Bit 6 gesetzt (wir zählen bei sowas immer von 0) -> sagt uns jetzt ohne weiten Hintergrund leider nichts ;-)

timbrd, 2024-10-17:
Kaum konfiguriert man den Splitter richtig, schon funktioniert alles. 😀

````
("WW Temperatur Oben (01CD)", 0x01CD, 3, 'b:0:1', 0.1, True),
("WW Temperatur Unten (01CE)", 0x01CE, 3, 'b:0:1', 0.1, True),
("WW Temperatur Mitte (01CF)", 0x01CF, 3, 'b:0:1', 0.1, True),
("WW Temperatur Ausgang (01D0)", 7720, 3, 'b:0:1', 0.1, True),
("WW Temperatur Soll (6000)", 0x6000, 3, 'b:0:1', 0.1, True),
````

Danke @philippoo66 und @FransOv für die Unterstützung.

@philippoo66 Deine sehr ausführliche und klare Erläuterung zur Nutzung des Bytebit Filters könnte man fast 1:1 in das Wiki stellen.

closed 2024-10-17 by timbrd

philippoo66, 2024-10-17:
ich hab's einfach mal rüber kopiert. kost ja nix und vlt hilft's.