philippoo66/optolink-splitter #52: Bit byte filter unexpected result

FransOv, 2024-11-16:
There are several 3 byte values where the first two bytes are a sensor value and the third byte is the sensor status.
When the sensor status is 00, 0xB40A, 3, 0.1, 'b:0:1', True gives a correct value, however when the sensor status (third byte) is non-zero the value is not correct.
r;0xB40a;3 => 1;0xB40A;9B0200
r;0xB40a;3;0.1;'b:0:1 => 1;0xB40A;66.4
r;0xB40d;3 => 1;0xB40D;000006 
r;0xB40d;3;0.1;'b:0:1 => 1;0xB40D;39321.6
It looks like all three bytes are still used: 393216 = 0x060000
I tested this via MQTT reads, don't know if it also has the same behavior in the polling list.

philippoo66, 2024-11-16:
Hi Frans,

`r;0xB40a;3;0.1;'b:0:1 => 1;0xB40A;66.4` there is a small little ' missing to finlize the bit-byte filter. please try
`r;0xB40a;3;0.1;'b:0:1'` and let me know

FransOv, 2024-11-16:
Missing single quote was a copy paste error.
r;0xB40a;3;'b:0:1';0.1 => 1;0xB40A;665
Third byte is 0, does not scale
r;0xB40a;3;0.1;'b:0:1 => 1;0xB40A;66.0
Does scale but possibly no byte select.

When thitd byte is 0x06
r;0xB40d;3;'b:0:1';0.1 => is rejected, no output
r;0xB40d;3;0.1;'b:0:1 => 1;0xB40D;39321.6
Does scale, but three bytes selected: 0x060000

philippoo66, 2024-11-16:
ok, I will have a ddeeper look. one moment or two please ;-)

FransOv, 2024-11-16:
I checked, same problem with query via the polling list. No need to hurry. It only gives problems when the sensor is not installed or is faulty.

philippoo66, 2024-11-16:
just to make sure: with the polling list it was always working, wasn't it? or does `...,"b0:1",0.1` not work in case rd byte is unequal 0?

philippoo66, 2024-11-16:
ok, fixed. it was a type error with scale factor when perform_bytebit_filter() was called from MQTT.

tested with `r; 0x2500; 22; 'b:1:2'; 0.05`
returned `1;0x2500;3251.3`
when 0x2500 was `02 02 fe d8 44 00 38 01 01 00 00 01 b4 00 00 00 00 38 01 00 b4 00`

so b:1:2 -> 02 fe -> 0xfe02 -> 65026
x 0.05 -> 3251.3

and fourth byte unequal 0

pushed to develop, merged into main.

philippoo66, 2024-11-16:
thank you for reporting, again! @FransOv

FransOv, 2024-11-16:
Thanks, it's working correctly now. My earlier remark that it had the same problem with polling was **not** correct. I made a type in the polling string.

closed 2024-11-16 by FransOv