philippoo66/optolink-splitter #54: Probleme bei Vitocal 300-A AWO-AC 301.B mit Vitotronic 200 WO1C

JackAAnders, 2024-11-26:
### Discussed in https://github.com/philippoo66/optolink-splitter/discussions/53

<div type='discussions-op-text'>

<sup>Originally posted by **JackAAnders** November 26, 2024</sup>
Hi,
der Optolink-Splitter will sich einfach nicht mit der Vitotronic verbinden. Bricht immer mit einem "VS2 protocol not detected within timeout" ab. 
Auch mit Viessdata kommt man out of the box nicht an die Daten ran. Viessdata scheint aber zumindest die richtige Anlage zu identifizieren.
[image]

Was mach ich falsch?

Grüße

</div>

JackAAnders, 2024-11-26:
--------------------------------------
versuche COM Port zu öffnen...
Port opened: COM10
Ok
versuche F8 mit KW Protokoll zu lesen...
RX: 05 
TX: 01 F7 00 F8 04 
RX: 20 
RX: 4C 
RX: 04 
RX: 11 
RX: 20 4C 04 11 
erfolgreich, Ident_F8: 20 4C 04 11 
Moment, kurz warten...
versuche Initialisierungs-Sequenz aus ViessData 2.3 ...
Init300ProtocolA()...
TX: 04 
RX: 05 
TX: 16 00 00 
RX: 06 
verbunden, Protokoll: VS2_300
versuche F8 mit ermitteltem Protokoll zu lesen...
TX: 41 05 00 01 00 F8 04 02 
RX: 06 41 09 01 01 00 F8 04 20 4C 04 11 88 
erfolgreich, Ident_F8: 20 4C 04 11 
schliesse COM Port...
TX: 04 
Port closed.
Test beendet.
Bitte Output kopieren und an Phil geben! Danke!

philippoo66, 2024-11-28:
hi @JackAAnders !
sorry, grad erst gesehen... das sieht doch jetzt gut aus. gibt es immernoch Probleme?

philippoo66, 2024-11-28:
ps. die DP Liste vom Viessdata passt so erstmal ganz und garnicht zu ner Wärmepumpe. die müssten wir grundlegend anpassen....

philippoo66, 2024-11-28:
ich ahne dein output ich von voident, richtig? hast du beim Optolink Splitter den COM Port richtig gesetzt? auf was für einem System läuft der überhaupt?

JackAAnders, 2024-11-29:
Sorry für die Konfusion, habe nur die hälfte erzählt glaube ich.

Setup:
Optolink Splitter soll auf RasPI laufen und wirft dort die Fehlermeldung "VS2 protocol not ...".
Hab mit lsusb genau geschaut, dass ich auch den richtigen /dev/ttyUSB habe.

Da das nicht funktioniert hat habe ich dann versucht mit Viessdata, von einem Windows Rechner aus die Daten auszulesen. Dazu hab ich direkt per USB verbunden. Das hat nicht so wirklich gut funktioniert, da kam noch nicht mal bei der Außentemperatur eine richtiger Wert zugrück, obwohl es so aussieht als ob Viessdata die Anlage vom Protokoll her richtig erkennt.

Danach hab ich dann versucht, ob ich das Protokoll mal testen soll. Dabei ist das logfile entstanden.

Dass ich die Viessdata DP Liste so nicht gebrauchen kann war mir dann auch irgendwann klar. Allerdings komme ich mit den DP_Listen für meine WP nicht so einfach zu einem XML, dass Viessdata versteht.

**Wer lesen kann ist klar im Vorteil, wenn man den `port_vitoconnect = None` stellt dann tut der optolik splitter auch was er soll.**

FransOv, 2024-11-29:
Die settings_ini.py Datei in https://github.com/philippoo66/optolink-splitter/discussions/21#discussioncomment-10170225  hat einige Fehler. Die letzte version ist im Wiki: https://github.com/philippoo66/optolink-splitter/wiki/210-Home-Assitant-Integration-of-Optolink%E2%80%90Splitter
300-G ist eine Sole Wärmepumpe, 300-A ist Luft Wärmepumpe.

closed 2025-01-23 by philippoo66