philippoo66/optolink-splitter #58: TypeError: 'type' object is not subscriptable

VDelport, 2024-12-08:
Hi,

after running the command

python3 optolinkvs2_switch.py

I get the following error message:

File "optolinkvs2_switch.py", line 25, in
import optolinkvs2
File optolink-splitter-main/optolinkvs2.py", line 77, in
def read_datapoint_ext(addr:int, rdlen:int, ser:serial.Serial) -> tuple[int, int, bytearray]:
TypeError: 'type' object is not subscriptable

I'm using Python 3.6.9 on Linux. Which Python version should I use to run the code without errors?

philippoo66, 2024-12-08:
hi @VDelport 

Python >= 3.9 required. Sorry, I'll add this to Readme 

pls let me know if everythin ok then!

philippoo66, 2024-12-08:
@VDelport please re-clone the settings_ini.py or remove the block comment part (Vitodens)! The block comment inside the list seams not to work.

closed 2025-01-04 by philippoo66