philippoo66/optolink-splitter #68: 'bytes' object cannot be interpreted as an integer

silaskreuze2008, 2025-01-04:
I was using the Optolink splitter without the Vitoconnect because I didn't have the CP2102, and it worked.

Today I upgraded my system with the CP2102 from AZDelivery. I configured it in the settings_ini.py using /dev/ttyS0 and /dev/ttyAMA0 now I'm getting the error: 'bytes' object cannot be interpreted as an integer

I am using a Pi 3B+ with the Optolinksplitter for the KW protocol.

Full console output:

python3 optolink_switch.py

optolinkvs1
awaiting Vitoconnect...
Server listening on 0.0.0.0:65234
Broker granted the following QoS: 0
'bytes' object cannot be interpreted as an integer
exit close
closing serViCon
reset protocol
closing serViDev
cancel poll timer
exiting TCP/IP client
disconnect MQTT client

[settings_ini.txt](https://github.com/user-attachments/files/18307413/settings_ini.txt)

philippoo66, 2025-01-04:
hi @silaskreuze2008 !

as far as I remember the KW protocol never got tested with vitoconnect, so we have to do 'pioneer work' now ;-)

you set `log_vitoconnect = True` - that's fine! does the vitolog.txt contain anything?

silaskreuze2008, 2025-01-04:
No Vitolog is completely empty.

I just tried to set the port of the CP2102 to none again, and it seems to work again, so maybe it's something serial-related or something to do with my Pi. I enabled the serial connection, etc.

philippoo66, 2025-01-04:
I found someting in detect_f8_read() what might hav caued thy type err and tried to fix it. Could you please try the new viconn_util.py?!

philippoo66, 2025-01-04:
detect_f8_read() is more or less the only funnction which is not used when there is no Vitoconnect...

silaskreuze2008, 2025-01-04:
Seems better!?

Now I'm getting this error: module 'viconn_util' has no attribute 'wait_for_vicon'

philippoo66, 2025-01-04:
I'll check...

silaskreuze2008, 2025-01-04:
I just replaced the complete file with the version from the main branch....

philippoo66, 2025-01-04:
main branch is wrong! you need to takte the viconn_util from vs1test branch

philippoo66, 2025-01-04:
in fact viconn_util from main branch has no wait_for_vicon(). there it is called detect_vs2()... ;-)

silaskreuze2008, 2025-01-04:
Now im oinly getting any further than this:

optolinkvs1
awaiting Vitoconnect...
Server listening on 0.0.0.0:65234
Broker granted the following QoS: 0
('Vitoconnect not detected within timeout', 120)
exit close
closing serViCon
reset protocol
closing serViDev
cancel poll timer
exiting TCP/IP client
disconnect MQTT client
closing vitolog

[vitolog.txt](https://github.com/user-attachments/files/18308568/vitolog.txt)

silaskreuze2008, 2025-01-04:
This is my [CP2102](https://www.amazon.de/dp/B089QJZ51Z?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1)

philippoo66, 2025-01-04:
hm, did you regard the sequence when starting?

philippoo66, 2025-01-04:
nothing is coming from the Vitoconnect...

philippoo66, 2025-01-04:
did you cross tx and rx when connecting the CP2102 board?

silaskreuze2008, 2025-01-04:
> did you cross tx and rx when connecting the CP2102 board?

Yes, I did. I have my Vito connected to a USB extension. I'm going to try if that's the problem.

Right start sequence is:
1. Start the script and wait for awaiting Vitoconnect...
2. Then turn on the VicoConnect.
Right?

philippoo66, 2025-01-04:
that's right.

by the way - last week the Vitonnect thing stopped working at Viessmann site, ref https://github.com/philippoo66/optolink-splitter/discussions/46#discussioncomment-11680317 I'm not sure if it 'returned' already..

silaskreuze2008, 2025-01-04:
Viessmann servers seem to be working now...

It worked! at least kind of.
The Vitoconnect connected, and I could control my heating unit, but then the script timed out.

[vitolog.txt](https://github.com/user-attachments/files/18308636/vitolog.txt)

philippoo66, 2025-01-05:
hm, timed out? the final line in the log says the opto device responded (1 byte, seems to be common with vs1). after that nothing more. I'm not sure what caused / what happened. wait, I'll analyse the traffic. vs1 I can't read just by heart...

philippoo66, 2025-01-05:
ok, master (Viconn) says read addr 5706, 1 byte. slave (opto) respons 55. seems to be right. I don't know why after that the Viconn does not continue

philippoo66, 2025-01-05:
5706 is _KonfiKTMaxGWG_, length 1 byte. So everthing correct so far.

silaskreuze2008, 2025-01-05:
let my try again

silaskreuze2008, 2025-01-05:
Again im only getting:

optolinkvs1
awaiting Vitoconnect...
Server listening on 0.0.0.0:65234
Broker granted the following QoS: 0
('Vitoconnect not detected within timeout', 120)
exit close
closing serViCon
reset protocol
closing serViDev
cancel poll timer
exiting TCP/IP client
disconnect MQTT client
closing vitolog

but while its running the vitoconnect is working fine.

[vitolog.txt](https://github.com/user-attachments/files/18308703/vitolog.txt)

philippoo66, 2025-01-05:
hm, communication is running well, but does not start with reading f8 (device ident). 

perhaps you need to stop everything, wait for some minutes and after that start again regrading the sequence?!

philippoo66, 2025-01-05:
without waiting some time, it is possible that Viessmann just think it's some timeout and continues afterwards (after restarting the script).

Up to now I've no idea how to detect the connection other that detecting the reading of f8 which alssways happens first usually...

silaskreuze2008, 2025-01-05:
also with the Opto 2?

philippoo66, 2025-01-05:
f8 always gets read initially. let me check if I have another idea detection communication even if reading something else

silaskreuze2008, 2025-01-05:
do you need more logs?

philippoo66, 2025-01-05:
I changed the detection to a 'not too safe way'. could you try?! viconn_util of vs1test branch...

philippoo66, 2025-01-05:
now the wait_for_vicon() returns true as soon as the response is of the demanded length....

philippoo66, 2025-01-05:
are you still awake and 'in action'? otherwise I'd go to bed now. tomorrow is another day... ;-)

silaskreuze2008, 2025-01-05:
Im stil in Action but i wanted to Go to bed also...

silaskreuze2008, 2025-01-05:
The Script ist running fine but the vitoconnect doesbt want to Connect.

When i start the skript it pauses then i Turn on the vitoconnect Script completes startup but the vitoconnect doesbt Connect

philippoo66, 2025-01-05:
ok, vitolog, please!?

silaskreuze2008, 2025-01-05:
Sorry one sec

silaskreuze2008, 2025-01-05:
[vitolog.txt](https://github.com/user-attachments/files/18308850/vitolog.txt)

philippoo66, 2025-01-05:
oh, now we are on vs2 again!

let me check

also please delete the old vitolog befor starting again. makes it easier to find the beginning ;-)

silaskreuze2008, 2025-01-05:
ok will do

philippoo66, 2025-01-05:
strange! now the master reads device ident f8 via vs1 and after that it tries to switch to vs2 but the opto does not 'speak' vs2

silaskreuze2008, 2025-01-05:
[vitolog.txt](https://github.com/user-attachments/files/18308859/vitolog.txt)

philippoo66, 2025-01-05:
the same. Vitoconnect tries to switch to vs2 protocoll. I don't understand. 20 94 is Vitotronic 200 KW1

philippoo66, 2025-01-05:
why didn't it do before?

silaskreuze2008, 2025-01-05:
> why didn't it do before?

I will try to connect the Vito Connect to my heating unit directly again and then switch back to the splitter.

philippoo66, 2025-01-05:
ok.

philippoo66, 2025-01-05:
there is another thing we could do: _serlog.py_.   This does nothing than forward every received byte (bidirectionally) and logs everything to file. this way we can see what the communication goes without the splitter script. and then try to find an according solution. 

but perhaps we should do that tomorrow?! ;-)

silaskreuze2008, 2025-01-05:
I think Tomorrow is better ;-)

[vitolog.txt](https://github.com/user-attachments/files/18308894/vitolog.txt)

philippoo66, 2025-01-05:
same thing - Viconn tries to change to vs2.

ok, let's continue tomorrow. It would be great if you could connect everything as with the splitter but run the serlog.py script and the start the Vitoconnect and after a few minutes stop it and upload the serlog_***.txt

silaskreuze2008, 2025-01-05:
will do

silaskreuze2008, 2025-01-05:
[serlog_250105025518.txt](https://github.com/user-attachments/files/18308911/serlog_250105025518.txt)

philippoo66, 2025-01-05:
funny! now it does not try to channge to vs2. let me sleep and think about this and what is the difference and how I can avoid that.

sleep well for today! ;-)

philippoo66, 2025-01-05:
good morning!

so what we see in serlog is

1. opto device regularly (~2s) posts 05h as usual
2. viconn tries to init vs2 - no success
3. viconn reads device ident using the very old GWG (!) protocol, first request initiated with leading 01h
4. vicon reads 0810h (TiefpassTemperaturwert ff. ??) with a length of 55 bytes (37h) using the vs1 protocol, responded correctly by opto dev
from then on it is continued using vs1.

I will try to write some code which applies to that detecting the viconn connection....

philippoo66, 2025-01-05:
ok, done. please try with new viconn_util of vs1test branch!

since vs1 reception in the script is also done in full raw mode, pleas also decrease full raw timings to
```
# full raw timing
fullraw_eot_time = 0.007    # seconds. time no receive to decide end of telegram 
fullraw_timeout = 1        # seconds. timeout, return in any case 
```

silaskreuze2008, 2025-01-05:
without the timing, it crashes viconnn turned off:

optolinkvs1
Server listening on 0.0.0.0:65234
awaiting Vitoconnect...
Broker granted the following QoS: 0
list index out of range
exit close
closing serViCon
reset protocol
closing serViDev
cancel poll timer
exiting TCP/IP client
disconnect MQTT client
closing vitolog

philippoo66, 2025-01-05:
ok, I fixed that (hopefully). please try again with new viconn_util ;-)

philippoo66, 2025-01-05:
I realized that sometimes, after some break (eg. line 476 ff.) the viconn switches back to GWG. but I hope due to raw mode this should not harm...

silaskreuze2008, 2025-01-05:
It starts detecting the Viconn, then gives me the normal log to the terminal, but Viconn doesn't connect, also getting some RX timeouts in the terminal output.

[vitolog.txt](https://github.com/user-attachments/files/18310239/vitolog.txt)

philippoo66, 2025-01-05:
I check...

silaskreuze2008, 2025-01-05:
[serlog_250105141110.txt](https://github.com/user-attachments/files/18310330/serlog_250105141110.txt)

philippoo66, 2025-01-05:
this again is like last time serlog. I will check if I can find a difference in timing...

philippoo66, 2025-01-05:
ok, perhaps the starting of the listening task takes too long. I will start that before detecting connection and clear the que when connection detected. hold on a minute or two... ;-)

philippoo66, 2025-01-05:
unfortunately this does ot work since both tasks access the serials :-/ let me think further...

[164 more replies]