philippoo66/optolink-splitter #70: Verbindung zu Optolink-Schnittstelle bricht ab

Skywalk0815, 2025-01-08:
Hallo zusammen,
beim Betrieb des Optolink-Splitters auf einen Raspi 5 kommt es in unregelmäßigen Abständen (von wenige Minuten bis zu einem Tag) zu einem Verbindungsabbruch der Optolink-Schnittstelle was zu folgendem Fehler führt:

[image]

[image]

Wenn der Optolink-Splitter nicht läuft bleibt die Optolink-Schnittstelle über Tage (bisher ca. 7 Tage getestet) verbunden.
Auch wenn der Splitter als Dienst läuft kommt es zu o.g. Fehler. Der Dienst behält trotzdem den Status "running".
Gibt es eine Möglickeit um den Fehler zu entdecken und den Dienst entsprechend neuzustarten?

Viele Grüße
skywalk

philippoo66, 2025-01-08:
moin @Skywalk0815 !

Es handelt sich hier um ein Problem mit der SW UART am USB wo der Optolink Kopf dran hängt (serViDev), oder mindestens mit der. Der serialposix bekommt halt nix mehr rausgeschrieben. 

Du bist übrigens nicht der Einzige, bei dem das Problem auftaucht, siehe hier https://github.com/philippoo66/optolink-splitter/discussions/46#discussioncomment-11452635 aber ihr beide seid die Ausnahmen, bei vielen anderen läuft es problemlos über Wochen und Monate durch (bei mir auch).

Ich habe leider keine Idee, was das verursacht. Immer wichtig ist eine gute und stabile Spannungsversorgung, aber es wäre schon zumindest bemerkenswert, wenn ein Mangel hier auf unterschiedlichen Systemen zu dem gleichen Fehler führt. Läuft auf dem Raspi sonst noch was, was zu genau dem Zeitpunkt irgendwas 'kritisches' macht?

Eigentlich sollte, da das Skript ja abbricht, dieses wieder neu gestartet werden `Restart=always` Eventuell könnte es helfen, die Wartezeit ewas hochzusetzen `RestartSec=...`

philippoo66, 2025-01-08:
ps. vielleicht kannst du mal gucken, ob du irgendwas mit Energiesparen für die USBs findest und das disablen. Wäre zwar komisch, weil das ja nun aktiv ist, aber wer weiss...

Skywalk0815, 2025-01-08:
Hi Phil,
vielen Dank für deine Hinweise.
Habe nun mal die Wartezeit auf 10s gesetzt. 
Die Spannungsversorgung vom Raspi ist sogar ein originales Netzteil.
Die grafische Oberfläche ist nicht aktiv und sonst ist das Raspi OS bisher nur mit den für den Splitter notwendigen Modulen ausgestattet. Als MQTT-Server läuft der Mosquitto. Den könnte ich aber zum Testen auch noch auslagern.
Einen aktiven USB-Hub werde ich heute Abend noch versuchen...

Skywalk0815, 2025-01-09:
Hey Phil,
den Test mit dem aktiven USB-Hub konnte ich noch nicht durchführen.
Aktuell habe ich mir beholfen die Zeile #295 serViDev.write([0x04])  auszukommentieren.
Der Splitter versucht noch etwas auf die Schnittstelle zu schreiben, obwohl davor schon nicht mehr geschrieben werden konnte.

philippoo66, 2025-01-09:
das macht aber nichts, ausser dass es eine (zusätzliche) Meldung gibt. Es ist besser, wenn es drin ist, weil damit beim Beenden das Protokoll zurückgesetzt wird, _wenn_ noch geschrieben werden kann.

Das eigentliche Problem ist ja, dass im Betrieb plötzlich nichts mehr rausgeschrieben werden kann. da hilft das Auskommentieren des Zurücksetzens beim Beenden auch nichts ;-)

Ich glaube nicht, dass die Stromaufnahme vom OptoKopf ein Problem ist (oder was meinst du mit 'aktivem' USB Hub?).

Skywalk0815, 2025-01-10:
Da hast du Recht, das eigentliche Problem liegt an einer anderen Stelle.
Allerdings wird der Dienst nicht neu gestartet wenn die Zeile #295 aktiv ist.
Wenn die Zeile auskommentriert ist wird der Dienst immer wieder neu gestartet. Das läuft jetzt auch schon 1,5 Tage am Stück so durch...
Vielleicht sollte ich noch eine Abfrage einbauen, ob das Script normal beendet wird oder über die Exception...

Die Sache mit dem USB-Hub hat, wie du schon angenommen hast, nicht funktioniert.
Vielleicht setze ich den Pi nochmal komplett neu auf.

philippoo66, 2025-01-10:
> der Dienst nicht neu gestartet wenn die Zeile #295 aktiv ist.

das ist interessant! hätte ich nicht vermutet, dass das einen Unterschied macht. 
dann macht das Auskommentieren ja wohl doch sinn... ;-)

philippoo66, 2025-01-10:
ich habe den exception Block im develop branh geänder zu 

```
    except Exception as e:
        excptn = e
        if(excptn == KeyboardInterrupt):
            print("Abbruch durch Benutzer.")
        else:
            print(e)
    finally:
        # sauber beenden: Tasks stoppen, VS1 Protokoll aktivieren(?), alle Verbindungen trennen
        # Schließen der seriellen Schnittstellen, Ausgabedatei, PollTimer, 
        print("exit close")
        if(serViCon is not None):
            print("closing serViCon")
            serViCon.close()
        if(serViDev is not None):
            if(serViDev.is_open and (excptn != serial.SerialException)):
                print("reset protocol")
                serViDev.write([0x04])  #TODO yes or no?
            print("closing serViDev")
            serViDev.close()
        print("cancel poll timer ") 
        timer_pollinterval.cancel()
        tcpip_util.exit_tcpip()
        #tcp_thread.join()  #TODO ??
        if(mod_mqtt_util is not None):
            mod_mqtt_util.exit_mqtt()
        if(vitolog is not None):
            print("closing vitolog")
            vitolog.close()
```

wäre lieb wenn du das testen würdest!

Skywalk0815, 2025-01-11:
Hi Phil,
das Script an sich läuft. Allerdings läuft das Script beim Abbruch der Verbindung zum Optolink trotzdem in den If-Zweig und will das Protokoll resetten.

philippoo66, 2025-01-11:
hm, das ist seltsam. ich schau noch mal.

philippoo66, 2025-01-11:
was gibt er denn bei dem  print(e) aus?

Skywalk0815, 2025-01-11:
Eigentlich das selbe wie zuvor:

Jan 11 10:29:37 raspberrypi python[17190]: OSError: [Errno 5] Input/output error
Jan 11 10:29:37 raspberrypi python[17190]: During handling of the above exception, another exception occurred:
Jan 11 10:29:37 raspberrypi python[17190]: Traceback (most recent call last):
Jan 11 10:29:37 raspberrypi python[17190]:   File "/home/markus/optolink-splitter-main/optolinkvs2_switch.py", line 314, in <module>
Jan 11 10:29:37 raspberrypi python[17190]:     main()
Jan 11 10:29:37 raspberrypi python[17190]:   File "/home/markus/optolink-splitter-main/optolinkvs2_switch.py", line 299, in main
Jan 11 10:29:37 raspberrypi python[17190]:     serViDev.write([0x04])
Jan 11 10:29:37 raspberrypi python[17190]:   File "/home/markus/optolink-splitter-main/myvenv/lib/python3.11/site-packages/serial/serialposix.py", line 655, i>
Jan 11 10:29:37 raspberrypi python[17190]:     raise SerialException('write failed: {}'.format(e))
Jan 11 10:29:37 raspberrypi python[17190]: serial.serialutil.SerialException: write failed: [Errno 5] Input/output error

philippoo66, 2025-01-11:
hm, der `OSError: [Errno 5] Input/output error` war mir tatsächlich durch die Lappen gegangen. Ich hab das angepasst, ist im develop hochgeladen.

Ist schon komisch, dass das unterschiedliche Exceptions gibt, wobel `serial.SerialException` ein 'Subtype' von OSError ist.

Chat GPT sagt dazu übrigens

[image]

scheint mir alles nicht so richtig wahrscheinlich, aber vielleicht fällt dir ja was dazu ein?

philippoo66, 2025-01-11:
vielleicht das hier noch

[image]

philippoo66, 2025-01-11:
aber das haben wir auch alles schon mit @thiemjoerg durch. Mir scheint als würde das System irgendwann auf die Idee kommen, mal wieder zu schauen, was es so an USB gibt is und das dann mal grad re-organisiert oder so, wobei dann eine bestehende Verindung im Nirvana landet...

Skywalk0815, 2025-01-15:
Hi Phil,
habe das neue Script nun seit Sonntag laufen und der Dienst wird bei jedem Fehler neugestartet. Ich muss somit nicht mehr händisch eingreifen. Super :-)
Danke dir für deine Mühen!

philippoo66, 2025-01-15:
danke dir für's Ausprobieren und die Rückmeldung!

thiemjoerg, 2025-01-19:
Hallo zusammen.
Ich weiß nicht, ob das hilft oder noch relevant ist.
Vor ca. 1 Woche hatte die Vitoconnect wieder geblinkt und die App zeigte die Meldung, dass es derzeit technische Probleme gibt. Ich musste den Dienst neu starten.
VG, Jörg

closed 2025-01-23 by Skywalk0815