philippoo66/optolink-splitter #72: Optolink VS2 RX Timeout Probleme

lellem, 2025-01-08:
Hallo Phil,

bei mir kommt es bei Laufzeiten < 24 h zu RX Timeouts bei der Optolink VS2 Schnittstelle. Der Service läuft weiter, die Kommunikation vom Raspi zur Heizung läuft und der MQTT Broker wird auch weiterhin beliefert.

Hast du eine Idee woran es liegen könnte?

Im Einsatz habe ich einen Pi4 mit 2GB.

philippoo66, 2025-01-08:
ich habe das Gefühl, dass Viessmann öfter mal an der Kommunikation 'runschraubt'. Oft ist es wochenlang ruhig, dann melden sich lötzlich mehrere Leute gleichzeeitig, dass es Probleme gibt, dann sucht und forscht man, findet aber nichts, und nach ner Weile ist plötzlich wieder alles gut, ohne dass am Splitter irgendwas geändert wurde. 

Grundsätzlich heisst _rx timeout_ im Viconn listen task einfach nur, dass 2 Sekunden (oder wodrauf es in der settings_ini einngestellt ist) nichts vom Vitoconnect empfangen wurde. das muss kein Fehler sein.

Was passiert denn, wenn du dann mittels ViCare mal irgendeine Einstellung änderst (Raumsolltemp oder so)?

philippoo66, 2025-01-08:
gegebenenfalls musst du mal `log_vitoconnect = True` in der settings_ini setzen, dann können wir mal nachschauen.

philippoo66, 2025-01-10:
gut zu wissen, danke!
das heisst wir können den issue hier schliessen?!

closed 2025-01-10 by lellem