philippoo66/optolink-splitter #75: Case für den Optolink Splitter

kristian, 2025-01-13:
Hallo @philippoo66, ich habe ein case für den Optolink Splitter designed und auf Printables hochgeladen:

https://www.printables.com/model/1144565-raspberry-pi-3-b-sleeve-case-w-cp2102-holder-wall

Hängt bei mir im Keller an der Wand. Danke für das Projekt! Wenn Du willst kannst Du den Link gerne noch in die Anleitung mit aufnehmen.

lellem, 2025-01-13:
@kristian Sehr cool, 
Wäre ein Case für einen Pi4 auch möglich zu designen?

kristian, 2025-01-13:
@lellem ich habe leider keinen RPi4 zur Hand. Soweit ich weiß sollten RPi3 und RPi4 rein layout technisch kompatibel zueinander sein. Auch der halter für das CP2102 Board sollte passen. Das müsste man mal Probedrucken.

philippoo66, 2025-01-13:
supi Sache, schreib ich nachher mal in's ReadMe.

Pi3 und Pi4 unterscheiden sich leider - die LAN und USB Buchsen sind da seitenvertauscht....

philippoo66, 2025-01-13:
[image]

links 4, rechts 3. Wenn du magst mess ich mal die Abstände zum Rand (und den zwischen den USB Buchsen), damit sollte es glaubich gegessen sein

kristian, 2025-01-13:
Mhm ja, ich sehe es. Ich habe das Case von einem anderen Modell "geremixed" (d.h. das Design des Cases kam nicht von mir), sondern von einem anderen Modell auf Printables: https://www.printables.com/model/15876-raspberry-pi-3-sleeve-case

D.h. da müsste man zumindest die USB / Netzwerkbuchse tauschen. Das kann ich machen, leider habe ich keine Möglichkeit einen Testdruck zu machen um das zu verifizieren.

lellem, 2025-01-13:
Ich könnte es drucken lassen, testen und Rückmeldung geben.

philippoo66, 2025-01-23:
@kristian soll ich messen oder bekommst du die Daten/Masse auf anderem Weg?

philippoo66, 2025-01-26:
@kristian  ich hab das in das README im develop branch übernommen. der wird ja bald zum main werden. Wie sieht's aus mit dem 4er?!

kristian, 2025-01-26:
Hey @philippoo66, danke für die Anpassung! Ich habe das Printables Modell gerade um das Sleeve Case für den RPi 4 erweitert. Wie gesagt, man müsste es noch Probedrucken, da ich keinen RPi 4 zur Hand habe!

philippoo66, 2025-01-26:
@lellem wolltest du dir nich eins drucken?! ;-)

lellem, 2025-01-27:
Ich kann's mir drucken lassen.

lellem, 2025-01-27:
Wo finde ich den Download?

kristian, 2025-01-27:
Auf Printables https://www.printables.com/model/1144565-raspberry-pi-3-b-4-sleeve-case-w-cp2102-holder-wal

philippoo66, 2025-01-31:
@lellem und 4er schon im Kasten? kann ich den Link übernehmen? ;-)