philippoo66/optolink-splitter #81: Verbindungsabbruch

HHBvB, 2025-01-21:
Nachdem der Service ein paar Minuten gelaufen ist, brechen folgende MQTT-Verbindungen zum Volkszaehler ab:
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/AussenTemp
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/KesselTemp
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/SpeicherTemp

Der Status des Optolink-Services lautet wie folgt:
pi@raspberrypi:~ $ sudo systemctl status optolink
● optolink.service - Optolink Python Service
     Loaded: loaded (/etc/systemd/system/optolink.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2025-01-21 16:31:28 CET; 30min ago
   Main PID: 4609 (python3)
      Tasks: 4 (limit: 4915)
        CPU: 17.143s
     CGroup: /system.slice/optolink.service
             └─4609 /usr/bin/python3 /home/pi/optolink-splitter-main/optolinkvs2_switch.py

Jan 21 17:00:33 raspberrypi python3[4609]: unknown first byte Error
Jan 21 17:00:33 raspberrypi python3[4609]: Error do_poll_item 21, Addr 6500, RetCode 32, Data ?
Jan 21 17:00:33 raspberrypi python3[4609]: rx 05
Jan 21 17:00:33 raspberrypi python3[4609]: unknown first byte Error
Jan 21 17:00:33 raspberrypi python3[4609]: Error do_poll_item 22, Addr 6513, RetCode 32, Data ?
Jan 21 17:00:33 raspberrypi python3[4609]: rx 05
Jan 21 17:00:33 raspberrypi python3[4609]: unknown first byte Error
Jan 21 17:00:33 raspberrypi python3[4609]: Error do_poll_item 23, Addr 6515, RetCode 32, Data ?
Jan 21 17:00:33 raspberrypi python3[4609]: rx 05
Jan 21 17:00:33 raspberrypi python3[4609]: unknown first byte Error

Viessdata 2.4 und MQTT-Explorer laufen aber weiterhin.

Der Status des Optolink-Services, wenn alles funktioniert, lautet wie folgt:
pi@raspberrypi:~ $ sudo systemctl status optolink
● optolink.service - Optolink Python Service
     Loaded: loaded (/etc/systemd/system/optolink.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2025-01-21 17:02:27 CET; 3min 25s ago
   Main PID: 24127 (python3)
      Tasks: 4 (limit: 4915)
        CPU: 2.073s
     CGroup: /system.slice/optolink.service
             └─24127 /usr/bin/python3 /home/pi/optolink-splitter-main/optolinkvs2_switch.py

Jan 21 17:05:31 raspberrypi python3[24127]: rx 41 06 01 01 a3 8f 01 8b c6
Jan 21 17:05:31 raspberrypi python3[24127]: rx 06
Jan 21 17:05:31 raspberrypi python3[24127]: rx 41 06 01 01 55 d3 01 4e 7f
Jan 21 17:05:31 raspberrypi python3[24127]: rx 06
Jan 21 17:05:31 raspberrypi python3[24127]: rx 41 07 01 01 a1 52 02 e4 80 62
Jan 21 17:05:31 raspberrypi python3[24127]: rx 06
Jan 21 17:05:31 raspberrypi python3[24127]: rx 41 07 01 01 65 00 02 86 01 f7
Jan 21 17:05:31 raspberrypi python3[24127]: rx 06
Jan 21 17:05:31 raspberrypi python3[24127]: rx 41 06 01 01 65 13 01 01 82
Jan 21 17:05:31 raspberrypi python3[24127]: rx 06

Was kann ich tun?

philippoo66, 2025-01-21:
moin @HHBvB !

hilf mir bitte auf die Sprüge! Wie stehen der Volkszähler und der Splitter in Verbindung? Eigentlich publishen doch beide nur Topics/Daten zum MQTT Broker?! Das einzige, wo sie sich da in die Quere kommen könnten, wäre ein identischer client-id string bei der Anmeldung beim Broker. Das halte ich aber für unwahrscheinlich, dass der Volkszähler da auch `"olswitch"` verwendet?!

Der obere Output deutet darauf hin, dass während der Kommunikation die Verbindung auf der Heizung (dem Optolink Client) zurückgesetzt wurde: Sie sendet eine 0x05, was sie gewöhnlich (zyklisch im 2-Sekunden Takt und nach Empfang einer Reset-Aufforderung 0x04) tut, um zu signalisieren, dass sie bereit für die Initialisierung des VS2 Protokolls oder für den Empfang eines (initialen) VS1 Telegramms ist.

Ist es möglich, das da der Splitter doppelt gestartet wurde?

HHBvB, 2025-01-22:
**Moin Phil,

danke für Deine schnelle Antwort!**

Dass der Splitter doppelt gestartet wurde, glaube ich nicht (er läuft als Service auf dem RPI). Auch der Volkszaehler läuft als Service auf demselben RPI.
Ich verstehe das so, dass der RPI als Broker für den Splitter funktioniert, auf den der Volkszähler mittels folgender Defintion eines Meters zugreift:
{			//Aussentemperatur
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "d452ca00-8e99-11ee-a31b-1f17634330dc",
          "identifier": "", 
          "middleware": "http://localhost/middleware.php",
          "duplicates": 0
        }
     ],
      "protocol": "exec",
      "command": "mosquitto_sub -C 1 -t Vitodens/AussenTemp",
      "format": "",
      "use_local_time": true
    },

Meine Heizung (Vitodens 333 von 2005) wird von Viessdata 2.4.0.2 uber TCP/IP als VDensHO1_4 (ab Softwareindex 4) identifiziert und über das Protokoll VS2_300 angesprochen.

Ich starte (stoppe) den Splitter wie folgt:
[Unit]
Description=Optolink Python Service
After=network.target
[Service]
ExecStart=/usr/bin/python3 /home/pi/optolink-splitter-main/optolinkvs2_switch.py
WorkingDirectory=/home/pi/optolink-splitter-main
StandardOutput=inherit
StandardError=inherit
Restart=always
RestartSec=10s
User=pi
Group=pi
[Install]
WantedBy=multi-user.target

Beim heutigen Test bekam nach erfolgreichem Start der Volkszähler nach 15 Minuten keine Daten mehr:
pi@raspberrypi:~ $ sudo systemctl status optolink
● optolink.service - Optolink Python Service
     Loaded: loaded (/etc/systemd/system/optolink.service; enabled; vendor pres>
     Active: active (running) since Wed 2025-01-22 09:14:37 CET; 13min ago
   Main PID: 32324 (python3)
      Tasks: 4 (limit: 4915)
        CPU: 7.659s
     CGroup: /system.slice/optolink.service
             └─32324 /usr/bin/python3 /home/pi/optolink-splitter-main/optolinkv>

Jan 22 09:27:42 raspberrypi python3[32324]: rx 41 09 01 01 08 8a 04 f5 be 08 00>
Jan 22 09:27:42 raspberrypi python3[32324]: rx 06
Jan 22 09:27:42 raspberrypi python3[32324]: rx 41 09 01 01 08 a7 04 b2 76 69 02>
Jan 22 09:27:42 raspberrypi python3[32324]: rx 06
Jan 22 09:27:42 raspberrypi python3[32324]: rx 41 06 01 01 0a 10 01 01 24
Jan 22 09:27:42 raspberrypi python3[32324]: rx 06
Jan 22 09:27:42 raspberrypi python3[32324]: rx 41 06 03 01 0c 20 01 01 38
Jan 22 09:27:42 raspberrypi python3[32324]: Error Message 01
Jan 22 09:27:42 raspberrypi python3[32324]: Error do_poll_item 14, Addr 0C20, R>
Jan 22 09:27:42 raspberrypi python3[32324]: rx 06

Die Status-Meldung besagt zwar, dass der Service aktiv ist, aber der MQTT-Broker scheint "down" zu sein (mosquitto_sub -C 1 -t Vitodens/AussenTemp auf der Konsole eingegeben "hängt". Auch Viessdata kann keine Verbindung mehr herstellen.

Nach einem Restart funktioniert alles wieder:
[... 105 lines cut ...]
    ("Stellung Umschaltventil", 0x0A10, 1, 1, False),

    ("Ruecklauftemp_calcd", 0x0C20, 2, 0.01, False),
    ("Pumpenleistung", 0x0A3C, 1, 1, False),
    ("Volumenstrom", 0x0C24, 2, 0.1, False),  # eigentlich scale 1 aber fÃ¼r Viessdata Grafik

    ("KesselTemp_soll", 0x555A, 2, 0.1, False),
    ("BrennerLeistung", 0xA38F, 1, 0.5, False),
    ("BrennerModulation", 0x55D3, 1, 1, False),

    ("Status", 0xA152, 2, 1, False),
    ("SpeicherTemp_soll_akt", 0x6500, 2, 0.1, False),
    ("Speicherladepumpe", 0x6513, 1, 1, False),
    ("Zirkulationspumpe", 0x6515, 2, 1, False),
    # bis hierher meine Viessdata Tabelle --------

#    ("Frostgefahr, aktuelle RTS etc", 0x2500, 22, 'raw'),
    ("Frostgefahr, aktuelle RTS etc", 0x2500, 22, 'b:0:21::raw'),
    ("Frostgefahr", 0x2500, 22, 'b:16:16::raw'),
    ("RTS_akt", 0x2500, 22, 'b:12:13', 0.1, False),

    # # 1-wire
    # ("SpeicherTemp_oben", 0xFFFd),
    # ("RuecklaufTemp_Sensor", 0xFFF4),
]

philippoo66, 2025-01-22:
> Ich verstehe das so, dass der RPI als Broker für den Splitter funktioniert, auf den der Volkszähler mittels folgender Defintion eines Meters zugreift:

ich glaub hier geht was durcheinander. 

- Der Broker (wohl mosquitto, wenn du mit _mosquitto_sub_ arbeitest) ist ein Programm, das du auf dem PI oder auch auf irgendeinem anderen Rechner im Netz laufen lässt.  
- Der Volkszähler ist ein anderes Programm, das du auch auf dem PI oder auch auf irgendeinem anderen Rechner im Netz laufen lässt, und das seine Daten an den MQTT Broker publisht. 
- Der Splitter ist ein anderes Programm, das du auch auf dem PI oder auch auf irgendeinem anderen Rechner im Netz laufen lässt, und das seine Daten an den MQTT Broker publisht. Er heisst 'Splitter', weil er die Optolink(!) Kommunikation aufsplittet auf bis zu 4 Master (Vitoconnect, Poll Mechanismus, MQTT und TCP Requests).

Was macht der Volkszähler mit den Viessmann Daten?? Der liest doch eigentlich Daten eines Zählers aus?

[image]

ich habe ChatGPT befragt zu dem, was du machst, so wie ich es verstehe:

```
Kann der Volkszähler 'frmede Daten' (bsw die einer Heizungssteuerung) vom MQTT Broker abholen und visualisieren?

ChatGPT:

Ja, der Volkszähler kann "fremde Daten", beispielsweise von einer Heizungssteuerung, von einem MQTT-Broker abholen und in seine Visualisierung integrieren. Dafür gibt es jedoch keine direkte, native Unterstützung im Standardumfang des Volkszählers. Es gibt allerdings Ansätze, wie du dies umsetzen kannst:
Vorgehen zur Integration von MQTT-Daten in den Volkszähler:

    MQTT-Daten abgreifen und umwandeln:
        Verwende ein zusätzliches Tool wie Node-RED oder ein Skript in Python, um Daten vom MQTT-Broker abzuholen und in ein Format zu bringen, das der Volkszähler versteht.
        Das übliche Format für den Volkszähler ist JSON oder CSV, das über vzlogger oder eine direkte Datenbankanbindung eingespeist wird.

    Einspeisung in die Volkszähler-Datenbank:
        Du kannst die Daten von der Heizungssteuerung über ein passendes Middleware-Tool (wie Node-RED) in die MySQL-Datenbank des Volkszählers schreiben. Dazu legst du einen eigenen Kanal (UUID) in der Volkszähler-Datenbank an, der speziell für die Heizungsdaten reserviert ist.
        vzlogger kann ebenfalls genutzt werden, wenn du die Daten lokal verfügbar machst und über eine Konfigurationsdatei einbindest.

    Visualisierung:
        Sobald die Daten in der Volkszähler-Datenbank sind, werden sie wie alle anderen Energiedaten behandelt und können mit der Standard-Visualisierung (z. B. im Frontend des Volkszählers) angezeigt werden.
        Du kannst auch benutzerdefinierte Labels oder Einheiten für die Kanäle verwenden, um die Daten korrekt darzustellen (z. B. Temperatur, Heizleistung).

Alternativen:

Falls du MQTT-Daten direkt visualisieren möchtest, könntest du auch Tools wie Grafana verwenden, die mit MQTT-Brokern integriert werden können. Grafana bietet weitreichendere Visualisierungsoptionen, die ebenfalls für Heizungssteuerungsdaten geeignet sind.
Fazit:

Der Volkszähler kann zwar keine MQTT-Daten direkt abholen, aber mit Tools wie Node-RED, Python oder durch eine eigene Middleware kannst du die Daten in den Volkszähler einspeisen und dort visualisieren. Es erfordert jedoch einige Anpassungen und technische Integration.
```

Welche Middleware benutzt du hier? Deine Architektur verstehe ich nicht, und ich befürchte, es ist nicht so gedacht wie du es benutzen möchtest.

Der Befehl `mosquitto_sub -C 1 -t Vitodens/AussenTemp` bewirkt, dass der 'Datenpunkt' aboniert wird und das Abo nach einmaligem(!) Empfang wieder gekündigt wird.

Ich könnte es mir nur so vorstellen, dass `http://localhost/middleware.php` das immer wieder macht, und es dann aber irgendwann schief geht, weil MQTT anders gedacht ist: da wird ein Topic vom Consumer beim Broker aboniert, und das bekommt der Abonent dann dauerhaft (vom Broker weitergeleitet), jedes Mal wenn es vom Producer an den Broker ge-publish-t wurde.

Vielleicht empfindet der Mosquitto Broker das ständige Abonieren und wieder Kündigen als eine Fehlfunktion, macht das eine Weile mit, aber sagt dann "_Da muss ein Wackelkontakt sein, ich ignoriere das jetzt mal, bevor es mein System zu stark belastet und ich zu nix anderem mehr komme_"?!

HHBvB, 2025-01-23:
Moin Phil,
ich möchte die Außentemperatur, die Kesseltemperatur und die Warmwasserspeichertemperatur im Volkszähler (VZ) visualieren.
Das funktioniert, indem man im VZ Meter definiert, die regelmäßig mittels folgender exec-Befehle ausgelesen werden:
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/AussenTemp
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/KesselTemp
mosquitto_sub -h 192.168.xxx.xx -C 1 -t Vitodens/SpeicherTemp
ChatGPT irrt - mit der o.a. Meter-Definition kann der VZ sehr wohl MQTT-Daten abholen.
VZ und Splitter laufen auf demselben RPI.
Genauso nutze ich den VZ zur Visualierung der Leistung meiner Balkonkraftanlage:
mosquitto_sub -C 1 -t solar/xxxxxxxxxxx/0/power
Hier werden die MQTT-Daten der OpenDTU abgeholt (als Broker fungiert wiederum der RPI), allerdings ohne Verbindungsabbrüche.

[image]

HHBvB, 2025-01-23:
[image]

philippoo66, 2025-01-23:
ok, wenn das prizipiell funktioniert, müssen wir schauen, was im konkreten Fall daneben geht.

Der Splitter publish-t ja die Topics/Daten nur an den mosquitto. Auf diesem Weg ist eine _Rückwirkung_ auf die Optolink Kommunikation _ausgeschlossen_.

Die einzige Möglichkeit, von MQTT aus 'Einfluss' auf den Splitter zu nehmen, ist das Senden von [Commands](https://github.com/philippoo66/optolink-splitter/wiki/010-Command-Syntax#command-syntax-for-requests-via-mqtt-and-tcpip) auf den Topic `Vitodens/cmnd`.

Kannst du mal Addr 0C20 aus der Polling Liste rausnehmen, den gibt es nämlich bei deiner Steuerung nicht. Sind da noch mehr, die es nicht gibt? (wobei sowas eigentlich vom Splitter sauber gehandhabt wird)

Dann würde ich dich bitten, den Splitter versuchsweise nicht als Dienst zu starten, sondern an der Kommandozeile, und dann den Output um den Moment, wo es abbricht, hier zu posten (da laufen ja die ganzen rx Telegramme durch, leider nicht die zum Splitter gesendeten, das müssten wir nötigenfalls einbauen) 

Ist es möglich, dass VZ _irgendwas_ mit dem serial Port macht, an dem der Optolink Kopf hängt? Der Splitter öffnet den zwar im exclusive-Modus, aber ich hab schon mal bemerkt, dass diese 'Verriegelung' nicht in jedem Falle wirkt.

Nochmal zum besseren Verständnis: Wenn du in VZ lediglich das Abholen der Splitter Daten weg lässt, aber sonst alles genauso läuft (parallel) gibt es das Problem nicht? Und es taucht zuverlässig reproduzierbar auf, nur wenn du die 3 Topics 'in den VZ holst'?

HHBvB, 2025-01-24:
Splitter von Kommandozeile gestartet - hier die letzten Meldungen:
rx 06
rx 41 09 01 01 08 8a 04 12 c0 08 00 7b
rx 06
rx 41 09 01 01 08 a7 04 d1 d2 69 02 cc
rx 06
rx 41 06 01 01 0a 10 01 01 24
rx 06
rx 41 06 03 01 0c 20 01 01 38
Error Message 01
Error do_poll_item 14, Addr 0C20, RetCode 3, Data 01
rx 06
rx 41 06 01 01 0a 3c 01 29 78
rx 06
rx 41 06 03 01 0c 24 01 01 3c
Error Message 01
Error do_poll_item 16, Addr 0C24, RetCode 3, Data 01
rx 06
rx 41 07 01 01 55 5a 02 fd 01 b8
rx 06
rx 41 06 01 01 a3 8f 01 00 3b
rx 06
rx 41 06 01 01 55 d3 01 00 31
rx 06
rx 41 07 01 01 a1 52 02 b0 20 ce
rx 06
rx 41 07 01 01 65 00 02 32 00 a2
rx 06
rx 41 06 01 01 65 13 01 00 81
rx 06
rx 41 07 01 01 65 15 02 00 00 85
rx 06
rx 41 1b 01 01 25 00 16 02 02 b3 62 6c 00 fd 01 01 00 00 01 fa 00 00 00 00 fd 01 00 fa 00 cf
rx 05
unknown first byte Error
Error do_poll_item 0, Addr 088E, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 1, Addr 0800, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 2, Addr 0802, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 3, Addr 0804, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 4, Addr 0808, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 5, Addr 5525, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 6, Addr 5523, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 7, Addr 5527, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 8, Addr 0A82, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 9, Addr 0884, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 10, Addr 5738, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 11, Addr 088A, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 12, Addr 08A7, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 13, Addr 0A10, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 14, Addr 0C20, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 15, Addr 0A3C, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 16, Addr 0C24, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 17, Addr 555A, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 18, Addr A38F, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 19, Addr 55D3, RetCode 32, Data ?
rx 05
unknown first byte Error
Error do_poll_item 20, Addr A152, RetCode 32, Data ?

HHBvB, 2025-01-24:
Habe 0C20 und 0C24 auskommentiert und es läuft jetzt seit 30 Minuten ohne Verbindungsabbruch.
Könnte es nicht doch an der Fehlerhandhabung des Splitters gelegen haben?
Die einzigen Fehler, die jetzt noch an der Konsole gezeigt werden, treten auf, wenn ich Viessdata gestartet habe:
rx 41 09 01 01 08 a7 04 98 d5 69 02 96
rx 06
rx 41 06 03 01 91 28 01 01 c5
Error Message 01
rx 06
rx 41 06 03 01 92 5c 01 01 fa
Error Message 01
rx 06
rx 41 06 03 01 93 90 01 01 2f
Error Message 01
rx 06
rx 41 06 03 01 90 20 01 01 bc
Error Message 01
rx 06
rx 41 06 01 01 0a 10 01 01 24
rx 06
rx 41 06 03 01 0c 20 01 01 38
Error Message 01
rx 06
rx 41 06 01 01 0a 3c 01 29 78
rx 06
rx 41 06 03 01 0c 24 01 01 3c
Error Message 01
rx 06
rx 41 06 01 01 23 02 01 00 2e
rx 06

Diese Fehler führen aber bisher nicht zu einem Verbindungsabbruch. Ich vermute ihre Ursache in der vito_DP.xml. Auch hier werden die Adressen 0C20 und 0C24 abgefragt.
Weisst Du, wo ich eine vollständige Adressenliste für meine Steuerung zur Anpassung von settings_ini.py und vito_DP.xml herbekommen könnte?

philippoo66, 2025-01-24:
data in case of 03 - ErrMsg (as known):
    0x01: invalid address

https://github.com/philippoo66/optolink-splitter/wiki/010-Command-Syntax#return_code

schmeiss mal bitte alle Datenpunkte raus, die zu Fehlern führen! (sowohl aus der poll_list als auch aus der vito xml)

ich hab mal die DP_VDensHO1_4.txt angehängt

[DP_VDensHO1_4.txt](https://github.com/user-attachments/files/18540041/DP_VDensHO1_4.txt)

vergl. https://github.com/philippoo66/ViessData21#datapoints_readme

vielleicht macht @FransOv  ja auch die Excel Tab für deine Steuerung fertig?! ;-)

s. https://github.com/philippoo66/optolink-splitter/wiki/310-Parameter-Addresses

FransOv, 2025-01-24:
Welche Anlage betrifft es?

philippoo66, 2025-01-24:
DP_VDensHO1_4.txt alles andre sind Links

philippoo66, 2025-01-24:
sorry Blödsinn völlig falsch verstanden

VDensHO1_4 heisst die 'Software'

philippoo66, 2025-01-25:
sag mal @HHBvB die Zeile da 
`rx 41 1b 01 01 25 00 16 02 02 b3 62 6c 00 fd 01 01 00 00 01 fa 00 00 00 00 fd 01 00 fa 00 cf`
liest ja DP 2500h aus. Das ist der letzte Datenpunkt in der poll_list, oder?

Wie groß hast du das poll_intervall gewählt? Ist es möglicherweise so groß, dass die Steuerung 'in der Zwischenzeit' ein Protokoll-(Versions)-Reset macht?

FransOv, 2025-01-25:
> VDensHO1_4 heisst die 'Software' ("Model" in der Excel) [@FransOv](https://github.com/FransOv)

Der Tab VDensHO1_4 ist an die Excel hinzugefügt.

philippoo66, 2025-01-25:
top, danke!

HHBvB, 2025-01-25:
Wie schon geschrieben: nachdem ich 0C20 und 0C24 auskommentiert habe, läuft der Splitter ohne Verbindungsabbruch.

closed 2025-01-25 by HHBvB

philippoo66, 2025-01-25:
ok, danke. Interessant, dass der Versuch des Auslesens von nicht vorhandenen berechnete Rücklauftemperatur und Volumenstrom derartige Folgen bedingt.

Vielleicht nutzt dir die Excel ja trotzdem was ;-)