philippoo66/optolink-splitter #97: How to determine byte position in composed datapoints?

philippoo66, 2025-02-07:
e.g. with CU401B_S we have 19 1-byte values at addr. 0x1410:

[image]

so from what do we know which byte belongs to which value? 

**we are missing some info like 'BytePosition'**...

initially I thought perhaps EventTypeOrder of ecnEventTypeEventTypeGroupLink might be the key but this seems to be a dead end

[image]

FransOv, 2025-02-08:
I think comparing the result of 0x1410 from different systems in different states might help. This is from a Vitocal 300-G with two heating circuits HK1 and HK2 both heating at the time: 1;0x1410;00000300000000000000000000000101000100
The 3 in there is intriguing as all values should be either 0 or 1.

meks007, 2025-02-23:
i'm currently trying to get frost protection from 0x1410, this is what 1410 looks like right now with frost protection currently ON.
one heating circuit, turned on but not heating right now:

1;5136;000001000100000000000000000001000000

will be taking another value tomorrow when it's warmer.

meks007, 2025-02-24:
so this is it with frost protection on and off
000001000100000000000000000001000000 <-- on
000001000000000000000000000001000000 <-- off

byte 5 is frost prot for heating circuit 1

ChrisOertel, 2025-02-24:
Die Frage ist ja, wie man das sicher rausbekommt, was was ist und ob es generell gilt und auch noch Gerätespezifisch ist. Für meine vitoCal 222 hab ich mich für folgendes entschieden (Heizkreis 1), was zu Deiner Erkenntnis passt:

Ist der Frostschutz generell eingeschaltet => (4,"WP.HK1.FrostschutzEingeschaltet (1410_2)",0x1410,19,'b:2:2'),
Ist der Frostschutz jetzt gerade aufgrund der Außentemperatur aktiv => (4,"WP.HK1.FrostschutzAktiv (1410_4)",0x1410,19,'b:4:4'),

philippoo66, 2025-02-26:
finding the byte positions empirically might be a way BUT I think the position assignment _should_ be documented somewhere in the SQL/xml data, don't you think?

Mini-Disc, 2025-03-29:
WIeso wird die 1410 denn mit 19 Bytes ausgelesen ? Die 1410 ist nur 10 Bytes lang. Das Besondere an der 1410 ist, das hier einzelne Bits innerhalb eines Bytes verschiedene Parameter spezifieren zB.:

    ("WPR3_Geraetestatus_Frostschutz_HK1~0x1410", 0x1410, 10, 'b:4:4:1' ),	
    ("WPR3_Geraetestatus_Frostschutz_HK2~0x1410", 0x1410, 10, 'b:4:4:2' ),			
    ("WPR3_Geraetestatus_Frostschutz_HK3~0x1410", 0x1410, 10, 'b:4:4:4' ),

Informationen wie BlockLengt (hier 10), BytePosition (hier 4), ByteLengt (hier 1), BitPosition (hier 39 für HK1) und BitLengt (1) fehlen leider in einigen Dokumentationen.

philippoo66, 2025-03-30:
hilfe, ja, dadran hatte ich jetzt nicht gedacht, dass einzelne Bytes auch noch mehrere 'gelistete' Informationen per einzelner Bits enthalten können.... 

> fehlen leider in einigen Dokumentationen.

wieso, gibt es auch Dokumentationen, aus denen das hervorgeht?

Mini-Disc, 2025-03-30:
Ich nutze die Datei "ecnEventType.xml" aus dem Vitosoft Verzeichnis "/ServiceTool/MobileClient/Config". Die Datei lade ich in Excel ein. Es steht leider nicht alles drin aber es hilft schonmal ungemein.

philippoo66, 2025-03-31:
in der Excel im Wiki hat Frans ja die ecnEventType noch mit allen möglichen anderen Tabellen verknüft und somit rausgeholt, was geht. Die xml ist ja die Grundlage für Marus 'sarnau' Fritze's Skript, mit dessen Hilfe die ganzen DP_... .txt erstellt worden sind - aber auch da Fehlanzeige bezüglich der gesuchten Informationen :-/