philippoo66/optolink-splitter #98: Added MQTT Topics in HomeAssistant do not show any data

loiselqw, 2025-02-11:
Hi there,

optolink-splitter is really great. I managed it to get relevant data from my Vitocal 300g to my broker and could access the data via MQTT Explorer.

[image]

See my settings and data attached. 

[homeassistant_entities.json](https://github.com/user-attachments/files/18750108/homeassistant_entities.json)

[settings_ini.py.txt](https://github.com/user-attachments/files/18750124/settings_ini.py.txt)

If I now use the homeassistant_create_entities.py script, the newly added sensors are showing up in HA, but are not linked to any mqtt data - at least this is my suggestion.

[image]

Do you have any ideas, how I could get data in HA as well?

Thanks in adv stefan

EarlSneedSinclair, 2025-02-11:
Are you using the latest `homeassistant_create_entities.py`? Please update this file from the repository if it is not up to date.

loiselqw, 2025-02-11:
Hi, thx for prompt reply. 

Still the same - as well with latest homeassistant_create_entities.py. "old" values are connected to MQTT broker, but the newly added sensors xy_JAZ keep "unbekannt".

EarlSneedSinclair, 2025-02-11:
Use MQTT Explorer to check if your JAZ sensors are listed. Copy the value here so we can verify if it has been published correctly.

You can delete the value(s) and trigger a new publishing run by starting `homeassistant_create_entities.py`. Watch the script’s output for any errors or disconnect messages.

Background: The script publishes entities based on the information in the JSON file to the MQTT topic `homeassistant`. Home Assistant retrieves them from there and creates the corresponding entities.
Previous versions had issues with disconnects during publishing, and Home Assistant only accepts entities from online devices. This should now be fixed, but please keep an eye on it.

loiselqw, 2025-02-11:
Hi, here we go, pls find a screenshot and the configs attached:

[image]

JAZ_ges:

{"object_id": "vitocal_jaz_ges", "unique_id": "vitocal_jaz_ges", "device": {"identifiers": ["Vitotronic_WO1C"], "name": "Vitocal", "model": "Vitocal 300G", "manufacturer": "Viessmann"}, "availability_topic": "Vito/LWT", "state_topic": "Vito/jaz_ges", "name": "JAZ_ges", "unit_of_measurement": "x", "icon": "mdi:multiplication"}

JAZ_heiz:

{"object_id": "vitocal_jaz_heiz", "unique_id": "vitocal_jaz_heiz", "device": {"identifiers": ["Vitotronic_WO1C"], "name": "Vitocal", "model": "Vitocal 300G", "manufacturer": "Viessmann"}, "availability_topic": "Vito/LWT", "state_topic": "Vito/jaz_heiz", "name": "JAZ_Heiz", "unit_of_measurement": "x", "icon": "mdi:multiplication"}

JAZ_ww:

{"object_id": "vitocal_jaz_ww", "unique_id": "vitocal_jaz_ww", "device": {"identifiers": ["Vitotronic_WO1C"], "name": "Vitocal", "model": "Vitocal 300G", "manufacturer": "Viessmann"}, "availability_topic": "Vito/LWT", "state_topic": "Vito/jaz_ww", "name": "JAZ_WW", "unit_of_measurement": "x", "icon": "mdi:multiplication"}

COP: => this one works

{"object_id": "vitocal_cop", "unique_id": "vitocal_cop", "device": {"identifiers": ["Vitotronic_WO1C"], "name": "Vitocal", "model": "Vitocal 300G", "manufacturer": "Viessmann"}, "availability_topic": "Vito/LWT", "state_topic": "Vito/cop", "name": "COP", "unit_of_measurement": "x", "icon": "mdi:multiplication"}

EarlSneedSinclair, 2025-02-11:
While the entity in Home Assistant is displayed in black and not greyish, I assume it has been picked up correctly by Home Assistant.  

When I experimented with my configuration, I noticed that Home Assistant only displayed values when they were currently published. For you, with `poll_interval = 30` and `(60, "JAZ_ges", 0x1680, 1, 0.1, False)`, this results in **1800 seconds** before a new value is available. Have you waited this time?  

Please remember that if you change the `poll_cycle`, you must restart `optolink-splitter` or your service.

loiselqw, 2025-02-11:
Hi,

I played a bit around and removed for COP and JAZ Values, the poll-cycle of 60:

[image]

Result keeps the same - in HA JAZ Values are NOT updated. Even in "Vito" topic they get data:

[image]

EarlSneedSinclair, 2025-02-11:
Have you deleted the retained messages in the Homeassistant discovery topics?
Have you (then) restarted Home Assistent?
Have you triep to rename to something completely different (e.g. SCOP)?

FransOv, 2025-02-13:
You are using upper case topics in setting_ini.py. The create entities script generates all lowercase topics from the specified name. That is why you are not getting any values as the topics are case sensitive.

loiselqw, 2025-02-13:
Hi FransOv,

that was it - now it works. Thank you

philippoo66, 2025-02-13:
>  The create entities script generates all lowercase topics

is that wise? shouldn't we change that since obviously it causes potential for mistakes?!

@FransOv @EarlSneedSinclair

philippoo66, 2025-02-13:
also I realized `re.sub(r"[^0-9a-zA-Z]+", "_", entity["name"]).lower()`

what does it do exactly? is it because HA can not deal with it?

EarlSneedSinclair, 2025-02-13:
HA can handle numbers in entity names, even though they shouldn't start with one.

Currently, the regex replaces all characters (including spaces, dashes, and other special characters) except 0-9, A-Z, and a-z with underscores (_) and converts the string to lowercase.

Below is an example of the script's printout. The entity names are clear and readable before being converted into the more technical form of settings_ini.

```
List of generated datapoint IDs (settings_ini), created from HA entities (homeassistant_entities.json):
  DP-ID: aussentemp | HA-Entity: Aussentemp | Domain: sensor
  DP-ID: cop | HA-Entity: COP | Domain: sensor
  DP-ID: durchlauferhitzer | HA-Entity: Durchlauferhitzer | Domain: switch
  DP-ID: hk2_heizkreispumpe | HA-Entity: HK2 Heizkreispumpe | Domain: binary_sensor
  DP-ID: hk2_heizkreispumpe_betriebsstunden | HA-Entity: HK2 Heizkreispumpe Betriebsstunden | Domain: sensor
  DP-ID: hk2_heizkurve_neigung | HA-Entity: HK2 Heizkurve Neigung | Domain: number
```

Are there any restrictions on names in the poll_list, @philippoo66?
@FransOv is the original creator of this script, so let's wait for his thoughts on this.

FransOv, 2025-02-13:
Entity id's as well as object id's in HA should be lower case, so I would prefer to keep it the way it is now.

EarlSneedSinclair, 2025-02-13:
I just read up on MQTT topic best practices. It's also a best practice (not a rule) to use lowercase topic names with no special characters.

So I’d prefer to keep it that way and maybe update the comments in settings_ini to emphasize using lowercase poll items.

philippoo66, 2025-02-13:
> Are there any restrictions on names in the poll_list

up to now I did not discover any. certainly I avoided äöüß§$%& and stuff like that, I don't know...

using / causes a sub-tree, eg.
```
hk1/heizkurve/neigung
hk1/heizkurve/niveau
hk2/heizkurve/neigung
hk2/heizkurve/niveau
```
turns out to
```
- hk1
  - heizkurve
    - neigung
    - niveau
- hk2
  - heizkurve
    - neigung
    - niveau
```

only when using `mqtt_fstr = "{dpaddr:04X}_{dpname}"` this turns out 'funny'. better to use `mqtt_fstr = "{dpname}_{dpaddr:04X}"` in such a case.

philippoo66, 2025-02-13:
> best practice to use lowercase topic names

oh, my complete stuff seems to be no good practice....

> update the comments in settings_ini

not only in settings_ini then. should be pointed to even in Readme I think. The Vitodens templates would have to get adopted, too, BUT they are taken from DP_xy.txt, and in the Excel they are capitals as well...

On Wiki pg. 350 three of five poll_lists use capitals. One even dots. We should think carefully about it...

EarlSneedSinclair, 2025-02-13:
Did you deliberately skip "(not a rule)" when quoting me? 😄
I guess it's something to consider for compatibility, but as mentioned, it's not a standard.

philippoo66, 2025-02-13:
rule is not practice ;-) 

but as we saw, it causes issues as it is now. perhaps we might think about a command line switch for create_entities?! Anyway we have to point to that clearly(!) in Readme, no matter how we end up.

EarlSneedSinclair, 2025-02-13:
While you point out that the Excel file contains capital letters, I wouldn’t opt to change all of them to lowercase. Also, if the Optolink-Splitter automatically converts them to lowercase, it might cause confusion during debugging, as the poll item inputs wouldn’t match the displayed MQTT topics.

A simple notice to use lowercase and avoid special characters should be sufficient.

I will update create_entities later to print a notice reminding users to double-check spelling and casing, right above the conversion list in my example above.

philippoo66, 2025-02-13:
I'd like to have a command line switch `--topicsunchanged` anyway...

philippoo66, 2025-02-13:
ok, your are the master of this HA stuff, so we do as you say. 

@EarlSneedSinclair will you do that in Readme as well (as in cerate.py)?

EarlSneedSinclair, 2025-02-13:
I have upgraded homeassistant_create_entities.py. It now checks if entities in homeassistant_entities.json are present in poll_items (case-sensitive). The output is displayed in an overview table, and if there is a mismatch, a warning with recommendations is shown.

Please see my [Pull Request (link)](https://github.com/philippoo66/optolink-splitter/pull/102) with complete output examples.

```
List of generated datapoint IDs (settings_ini), created from HA entities (homeassistant_entities.json):

        |                     | Generated         | DP found     
Domain  | HA-Entity from json | Datapoint (DP)    | in poll_list?
climate (1) -----------------------------------------------------
        | HK2 Thermostat      | hk2_thermostat    | No           
sensor (3) ------------------------------------------------------
        | Aussentemp          | aussentemp        | Yes          
        | COP                 | cop               | Yes          
        | Resp                | resp              | No           
switch (1) ------------------------------------------------------
        | Durchlauferhitzer   | durchlauferhitzer | No           
text (1) --------------------------------------------------------
        | Cmnd                | cmnd              | No           
-----------------------------------------------------------------
TOTAL ENTITIES: 6

!!! WARNING !!!
One or more entities from your homeassistant_entities.json were not found in your poll_items (settings_ini.py or poll_list.py).

Possible causes:
 - Thermostats consist of multiple values with no direct counterpart in poll_items.
 - Switches consist of multiple values with no direct counterpart in poll_items.
 - "Cmnd" refers to commands sent from Home Assistant to the Optolink-Splitter, and therefore has no direct counterpart in poll_items.
 - "Resp" is the response to commands sent by the Optolink-Splitter, and therefore has no direct counterpart in poll_items.

For other entities, especially SENSORs, this could be critical, as they will not attach to a value and will not be displayed in Home Assistant.

To prevent this issue, PLEASE CHECK SPELLING AND ENSURE THAT POLL_ITEMS ARE IN LOWERCASE.
Recommendation: Keep all MQTT-related values (e.g. mqtt_topic) in lowercase!
```

FransOv, 2025-02-14:
Super!

philippoo66, 2025-02-14:
jap, top Sache! Hast du ja richtig Arbeit reingesteckt! sieht echt gut aus!

philippoo66, 2025-02-14:
ist gemerged. Danke!!

philippoo66, 2025-03-05:
können wir den Issue hier eigentlich closen?!

FransOv, 2025-03-05:
The original issue was already closed once  and the merge has been done, so yes, in my opinion it can be closed.

closed 2025-03-05 by philippoo66