philippoo66/optolink-splitter #107: LWT goes offline after each cycle

meks007, 2025-02-23:
There is a bug with the LWT feature introduced in #30 when using HA entities.
HA entites are created with an availability topic of vito/LWT. When LWT goes offline, HA entities go unavailable.

Now, Optolink Splitter connects and disconnects on each poll cycle, resulting in LWT continuously changing from online/offline every couple of seconds. This results in HA entities going available/unavailable too.

Either remove the availability topic from HA entites or don't drop the mqtt connection at the end of a poll cycle.

meks007, 2025-02-23:
Sorry, it appears as though there was a misconfiguration on my end, resulting in a continuous service restart.

closed 2025-02-23 by meks007

philippoo66, 2025-02-23:
ok, thank you for clarification. Are you using the current master branch (V1.2.0.0) or the new (1.3.0.0) in develop with built-in reconnect on Viconn issues? Since you say 'service restart' I think it is the V1.2, right?

meks007, 2025-02-23:
yep, current master branch, but develop sounds promising!