philippoo66/optolink-splitter #118: Recover from Timeout: RetCode 255

PhilippRaab, 2025-04-17:
Hey,

i have an issue with the optolink splitter. It runs fine for a couple of hours, but when it runs into a timeout once it never recovers. It reports RetCode 255 for every poll.
If i restart the script it works again immediately, so i think there is no problem with the connection.

Any idea what could be the issue?

[image]

Thanks for your work. BR, Philipp

philippoo66, 2025-04-17:
hm, funny! what is going on? The splitter was working since years without any problems on optolink comm and now we have it twice within 2 days @ 2 users?!?!

I will have a look at that also @PhilippRaab . Please also watch https://github.com/philippoo66/optolink-splitter/discussions/117#discussioncomment-12859701

philippoo66, 2025-04-18:
I checked - I think there _is_ a problem with the connection (communcation). If you restart the splitter, during init a communication reset gets sent to the vito. So if there was a problem, it gets 'solved' due to that. I am quite sure that there is no issue in coding - the timeout counter etc gets reinitalized each time on trying receiving a response to a sent request.

But I wonder what causes the issue. Could you pls describe your setup?! What 'target', with/wo Vitoconnect, what splitter version, did you change settings_ini.olbreath or any other settings, virtual environment, ...

philippoo66, 2025-04-18:
thx! as mentioned in the other issue thread, I'm thinking about implementing an error counter for the optolink comm and utilize the already implemented re-init loop in case of Viconn failure when the optolink-comm-error-counter reaches a certain threshold.

I'm still thinking because the MQTT connect is not 'inside' this re-init loop, and @Meramsonek told of MQTT problems. 

Change like this always bear the danger of implementing something 'unintended' and need carefully to get tested - including all options, e.g. Viconn, so I'm a bit hesitating...

closed 2026-02-12 by philippoo66