philippoo66/optolink-splitter #130: fehlerhaftes Poll Item führt zum Absturz

noseshimself, 2025-09-16:
... obwohl der Fehler erkennbar sein sollte.

Gegeben: eine CU401B_S und eine DP_Liste_2

CU401B_S	WPR_WW_Betriebsart~0x6004	0x6004	NoConversion	R/W	1	3		NULL	VarChar	NULL	NULL	NULL	0,1,2,3	STAND BY,NUR NACH SCHALTZEITEN,DREHSCHALTER,STANDARD

```
[...]
    (10, "Warmwasser/Betriebsart (RW)", 0x6004, 1, False), # STAND BY,NUR NACH SCHALTZEITEN,DREHSCHALTER,STANDARD
    (10, "Warmwasser/Warmwasser-Speicher reaction force (RW)", 0x600F, 'b:0:0:3', False), # 0,1,2 KEINE WÄRMEABNAHME,NUR BEI KRITISCHEM ÜBERSCHUSS,ZU JEDER ZEIT
    (10, "Warmwasser/Warmwasser-Vorrang", 0x6010, 1, 'onoff', False),
[...]
```
Ok, wenn also die Werte bei 600F nur 0, 1 oder 2 sein dürfen, bei der Abfrage aber beliebige höhere Werte gemeldet werden, gibt der Controller da etwas von sich, was (mal wieder) nicht dokumentiert ist, daher ein && 3, um das Ergebnis zu filtern.

Ergebnis:

```
[...]
rx 41 06 01 61 7e 82 01 00 69
rx 41 05 00 81 7e 83 01 88
rx 06
rx 41 06 03 01 60 04 01 01 70
Error Message 01
Error do_poll_item 146, Addr 6004, RetCode 3, Data 01
rx 06
rx 41 06 01 81 7e 83 01 04 8e
rx 41 05 00 a1 7e 91 02 b7
rx 06
rx 41 07 01 a1 7e 91 02 00 00 ba
rx 41 05 00 c1 7e 92 02 d8
invalid literal for int() with base 10: 'b:0:0:3'
exit close...
cancel poll timer
exiting TCP/IP client
closing serViCon
Error in listen_to_Vitoconnect: Error aa in receive_vs2telegr, data:  re-init
reset protocol
closing serViDev
disconnect MQTT client
closing vitolog
```

1) Soll das so?
2) Was mache ich mit data points, die mehr liefern, als in der Liste steht (das sind typischerweise diese Wertelisten, für die "ja/nein" nicht reicht; ich halte sie für einen Architekturfehler seitens Viessmann, da hätte man besser Bit-Arrays statt Aufzählungen genommen), um den Rest, den man nicht versteht, auzublenden?

philippoo66, 2025-09-17:
> (10, "Warmwasser/Warmwasser-Speicher reaction force (RW)", 0x600F, 'b:0:0:3', False), 

so, der Fehler hat diesmal nichts mit der Spannungsversogung zu tun, sondern die **Syntax ist falsch**. Sie lautet

[image]<br>

der Splitter versucht also den BB Fiter string als length zu interpretieren.

noseshimself, 2025-09-18:
> der Splitter versucht also den BB Filter string als length zu interpretieren, das muss schief gehen.

Guter Grund, aber dann ist die Fehlermeldung sehr unglücklich (bzw. die Stelle, an der es dann knallt).

> Aber trotzdem ist es wieder mal so eine Sache, wenn man nicht weiss, ob der Controller korrekt läuft oder wegen der Spannungsversorgung irgendwelche 'unlogischen' Sachen macht...

Keine Angst, der läuft jetzt seit etwas über einer CPU-Stunde stabil durch.

> und ja, das soll so. Das Skript soll exiten bei sowas, weil es keinen Sinn macht, den Fehler einfach abzufangen und bei jedem Zyklus wieder reinzutappen. Und dann kommt wieder die Frage "Warum wird mein Poll Item nicht gelesen?"

Schöner wäre dann ein anderer Fehler als das ominöse "AA" (es sei denn, das heißt "ich bin am Ar...").

Aber damit ist das Problem aus der Welt, das freut doch auch.