philippoo66/optolink-splitter #133: Feature recommendation: New API command "sp" (set poll_cycle)

noseshimself, 2025-09-24:
Ja, eigentlich ein Request, aber eher als Denkanstoß:

poll_cycle wird ja eh bei jedem Durchlauf ausgewertet. wenn man den mit

``sp:[data point]:[neuer Wert]``nachträglich ändern könnte, würde man auf "WP x ist an" mit "sp:[Energieverbrauch],1" reagieren und das hinterher wieder heruntersetzen. Das könnte zumindest bei Weiterverwendung per Home Assistant MQPP (statt NodeRed) deutlich Datenbank-Platz sparen.  (Im Moment muß ich meinen Energieverbrauch aus den Leistungsdaten selber integrieren, da sind kleinere Deltas schon hilfreich, ansonsten interessiert mich der Energieverbrauch der Wärmepumpe nicht die Bohne, also würde ich da auf die Werte gerne verzichten. Innerhalb von HA ist das... schwierig.)

philippoo66, 2026-02-04:
@noseshimself  hast du das ausprobiert? erfolgreich? 

ich schliesse immer gerne issues... ;-)

jaegerschnitzel, 2026-02-04:
> so Leute, es ist implementiert (branch develop). bitte testen!
> 
> https://github.com/philippoo66/optolink-splitter/wiki/010-Command-Syntax#poll_cycle

funktioniert wie es soll

noseshimself, 2026-02-05:
> [@noseshimself](https://github.com/noseshimself) hast du das ausprobiert? erfolgreich?
> 
> ich schliesse immer gerne issues... ;-)

Was soll die blöde Frage, natürlich funktioniert es direkt aus der Werkstatt. 8-) Was auch sonst?

Ich versuche gerade verzweifelt, eine KI so zu dressieren, daß sie meine Gedanken liest und austomatisiert aus einer etwas verbesserten poll_list die HA-Sensoden bauen kann. Aber eigentlich glaube ich, daß ich eher ein Tool bekomme, das ein Datenbank irgendeiner Art nimmt (ich weiß, Du haßt das Wort; also irgendeine Art strukturierter Datensammlung) und daraus sowohl poll_list als auch Sensoren generiert. google.ai fragt mich dauernd höflich, ob ich komplett meschugge sei, weil ich mir die falsche Wurzel für meine Abhängigkeiten suche (eben die poll_list) und python könnte der Idiot nicht mal, wenn sein Leben davon abhinge.

philippoo66, 2026-02-05:
ist nicht ha_shared_config und ha_publish das was du suchst?

@matthias-oe und @EarlSneedSinclair überarbeiten das grade noch ein wenig, wenn du da noch input zu hast...

oder willst du kein auto-ddiscovery sondern was, das dir den toml Kram baut?

noseshimself, 2026-02-05:
> ist nicht ha_shared_config und ha_publish das was du suchst?

Ich möchte das selber machen, da ich ja kein einzelnes flaches Device mit 8000 Sensoren habe, sondern einen ganzen Baum aufbaue.

closed 2026-02-05 by philippoo66